wasm-micro-runtime/wasm-micro-runtime #3394: Empty `fd_pread` fails when it should succeed

yagehu, 2024-05-06:
### Subject of the issue

Calling `fd_pread` with a 0-length `iovs` is a no-op on host Linux and other Wasm runtimes (Wasmtime, Wasmer, Node) while WAMR fails with `inval` (28).

### Test case

```
fn main() {
    unsafe {
        let fd = wasi::path_open(
            3,
            0,
            "a",
            wasi::OFLAGS_CREAT,
            wasi::RIGHTS_FD_SEEK | wasi::RIGHTS_FD_READ,
            0,
            0,
        )
        .unwrap();

        let iovs = [];

        wasi::fd_pread(fd, &iovs, 0).unwrap();
    }
}
```

### Your environment

Tested with HEAD on Linux and Darwin amd64.

### Steps to reproduce

Run the example above.

### Expected behavior

It should not panic and `fd_pread` into an empty buffer should be no-op.

### Actual behavior

The code panics with iwasm. But not other runtimes.

yamt, 2024-05-07:
> It should not panic and fd_pread into an empty buffer should be no-op.

i tend to disagree.
EINVAL for iovcnt==0 comes from posix. https://pubs.opengroup.org/onlinepubs/9699919799/functions/readv.html
wasm3 and toywasm fail with `inval`. (as i expect)
i feel it's a bug in the "other runtimes" you tested.

yagehu, 2024-05-07:
That's a good point. I was looking at Linux man pread and missed the POSIX requirement.

closed 2024-05-07 by yagehu