wasm-micro-runtime/wasm-micro-runtime #3401: Unexpected exception on faster interpreter mode

erxiaozhou, 2024-05-08:
### Subject of the issue
WAMR on fast interpreter mode raises an exception indicating "invalid local type" on a valid case.
### Test case

[f64.const_6LLPBLPP_856_1714940342309342.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15244364/f64.const_6LLPBLPP_856_1714940342309342.zip)

### Your environment

* Host OS: Ubuntu 20.04
* Commit Id: 1c2a8fca4e9bf969ae27240a0918ee8164dfd598

### Steps to reproduce
1. build WAMR on fast interpreter mode
2. execute the command
```
iwasm --heap-size=0 -f "to_test" <test_case>
```
### Expected behavior
No exception
```
-0.5392354:f64
```
### Actual behavior
An exception
```
Exception: invalid local type
```

wenyongh, 2024-05-14:
@erxiaozhou thanks for reporting the issue, please try #3434.

wenyongh, 2024-05-21:
Close the issue since it was fixed.

closed 2024-05-21 by wenyongh