wasm-micro-runtime/wasm-micro-runtime #3402: Unexpected behavior on fast interpreter mode

erxiaozhou, 2024-05-08:
### Subject of the issue
WAMR on fast interpreter mode raises an unexpected exception.
### Test case

[tt_finterp_exceptionx.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15244715/tt_finterp_exceptionx.zip)

### Your environment

* Host OS: Ubuntu 20.04
* Commit Id: 1c2a8fca4e9bf969ae27240a0918ee8164dfd598

### Steps to reproduce
1. build WAMR on fast interpreter mode
2. execute the command
```
iwasm --heap-size=0 -f "to_test" <test_case>
```
### Expected behavior
No exception
```
0x2d0:i64
```
### Actual behavior
An exception
```
Exception: integer overflow
```

wenyongh, 2024-05-14:
@erxiaozhou thanks for reporting the issue, please try #3434.

wenyongh, 2024-05-21:
Close the issue since it was fixed.

closed 2024-05-21 by wenyongh