wasm-micro-runtime/wasm-micro-runtime #3403: Wrong execution reuslt on fast interpreter mode

erxiaozhou, 2024-05-08:
### Subject of the issue
WAMR on fast interpreter mode  outputs a wrong result on the test case
### Test case

[finterp_wrong_val_select.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15246254/finterp_wrong_val_select.zip)
```
(module
  (type (;0;) (func))
  (type (;1;) (func (result i64)))
  (type (;2;) (func (param i32 i32 i32) (result i32)))
  (func (;0;) (type 1) (result i64)
    i64.const 122
    i32.const 1
    i32.const 2147483647
    i32.const -1
    i32.const 1
    if (param i32 i32 i32) (result i32)  ;; label = @1
      select
    else
      br 0 (;@1;)
      nop
      select
    end
    drop
    )
  (export "_start" (func 0))
  (export "to_test" (func 0)))

```

### Your environment

* Host OS: Ubuntu 20.04
* Commit Id: 1c2a8fca4e9bf969ae27240a0918ee8164dfd598

### Steps to reproduce
1. build WAMR on fast interpreter mode
2. execute the command
```
iwasm --heap-size=0 -f "to_test" <test_case>
```
### Expected behavior
The output is supposed to be the operand 122, which is introduced by the first instruction in the function.
```
122
```
### Actual behavior

```
0x100000001:i64
```

TianlongLiang, 2024-05-08:
Hi, I tested it on my Ubuntu 20.04 local machine, and it output the expected answer:
[image]

erxiaozhou, 2024-05-08:
> Hi, I tested it on my Ubuntu 20.04 local machine, and it output the expected answer

Thank you for your reply, you are right! I found the bug has been fixed by the version of commit id e44465d2599d5fc4bf66a6469202898c32b75edf .

closed 2024-05-08 by erxiaozhou