wasm-micro-runtime/wasm-micro-runtime #3407: "duplicate export name" check is O(n^2)

yamt, 2024-05-09:
"duplicate export name" check in load_export_section is too slow for modules with many exports.

wenyongh, 2024-05-09:
Maybe we can create the export list first, and then sort the list by name, and then check whether the contiguous two exports have the same name, and report error if yes?

yamt, 2024-05-09:
> Maybe we can create the export list first, and then sort the list by name, and then check whether the contiguous two exports have the same name, and report error if yes?

maybe.
for the other runtime, i do something along the line. https://github.com/yamt/toywasm/blob/115e895007bb77b12f6a632a3f4354644b236c56/lib/module.c#L2122-L2143

wenyongh, 2024-05-09:
OK, got it, thanks.

wenyongh, 2024-07-01:
@yamt I tried to refine the code to quick-search (bsearch) the wasm/aot export with name, by sorting the export list by name after loading the list. But it runs failed when running the wasm-c-api sample, I found that it is caused by:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/ea582fbc07468b2ef11a13d5ac18058047e50ef7/samples/wasm-c-api/src/callback.c#L157

This file was ported from:
https://github.com/WebAssembly/wasm-c-api/blob/main/example/callback.c#L133

In which it assumes that the export order should not be changed and `exports.data[0]` stores the related data of wasm file's export 0. So it seems that sorting export list with name is not a good way. How about creating an array of `<name, export_index>` pair and sort it by name instead?

yamt, 2024-07-01:
> @yamt I tried to refine the code to quick-search (bsearch) the wasm/aot export with name, by sorting the export list by name after loading the list. But it runs failed when running the wasm-c-api [...]

i tend to think it's a bug in the sample. but i agree probably it's safer to preserve the order.

> This file was ported from: https://github.com/WebAssembly/wasm-c-api/blob/main/example/callback.c#L133
>
> In which it assumes that the export order should not be changed and `exports.data[0]` stores [...]

it makes sense.