wasm-micro-runtime/wasm-micro-runtime #3410: iwasm: WASM module load failed: struct or array as field is not supported in constant expr

bashor, 2024-05-09:
### Steps to reproduce

1. Unpack zip [kotlin-wasm-wasi-example-wasm-wasi-prod.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15267820/kotlin-wasm-wasi-example-wasm-wasi-prod.wasm.zip)
2. Build `iwasm` with the following command
```
 cmake .. -DWAMR_BUILD_GC=1 -DWAMR_BUILD_BULK_MEMORY=1 -DWAMR_BUILD_EXCE_HANDLING=1 -DWAMR_BUILD_TAIL_CALL=1 -DWAMR_BUILD_CUSTOM_NAME_SECTION=1 -DWAMR_BUILD_EXCE_HANDLING=0 && make
```
3. Run `./iwasm /root/tmp/kotlin-wasm-wasi-example-wasm-wasi-prod.wasm`

### Expected behavior

It should print something like:
```
Hello from Kotlin via WASI
Current 'realtime' timestamp is: 1715281633864118000
Current 'monotonic' timestamp is: 210424654558083
```

### Actual behavior
It prints:
```
iwasm: WASM module load failed: struct or array as field is not supported in constant expr
```

TianlongLiang, 2024-05-13:
I took a look at this case, it's due to the [current implement limitation](https://github.com/bytecodealliance/wasm-micro-runtime/blob/22eef3d88dd7795a924e4d73f00dc4459b02e7e5/core/iwasm/interpreter/wasm_loader.c#L560), we don't allow to init the global with references. The #3411 is also due to this limitation.

bashor, 2024-05-13:
Any chance they will be fixed in the foreseeable future?

TianlongLiang, 2024-05-14:
We plan to take on that and will try our best to support this syntax, it could be a while through

bashor, 2024-05-19:
@wenyongh Hi! Thanks for the fix! Could you please also check it on #3411? It's an unoptimized version and should contain more cases.

wenyongh, 2024-05-20:
@bashor Welcome, I looked into this case and fixed several issues in wasm loader, now iwasm can run this case successfully, please try PR #3447 again.