wasm-micro-runtime/wasm-micro-runtime #3411: iwasm: WASM module load failed: type mismatch: expect (ref null ht) but got other1 unknown type

bashor, 2024-05-09:
### Steps to reproduce

1. Unpack zip [kotlin-wasm-wasi-example-wasm-wasi-dev.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15267851/kotlin-wasm-wasi-example-wasm-wasi-dev.wasm.zip)

2. Build `iwasm` with the following command
```
 cmake .. -DWAMR_BUILD_GC=1 -DWAMR_BUILD_BULK_MEMORY=1 -DWAMR_BUILD_EXCE_HANDLING=1 -DWAMR_BUILD_TAIL_CALL=1 -DWAMR_BUILD_CUSTOM_NAME_SECTION=1 -DWAMR_BUILD_EXCE_HANDLING=0 && make
```
3. Run `./iwasm kotlin-wasm-wasi-example-wasm-wasi-dev.wasm`

### Expected behavior

It should print something like:
```
Hello from Kotlin via WASI
Current 'realtime' timestamp is: 1715281633864118000
Current 'monotonic' timestamp is: 210424654558083
```

### Actual behavior
It prints:
```
WASM module load failed: type mismatch: expect (ref null ht) but got other1 unknown type
```

TianlongLiang, 2024-05-10:
Hi, would you mind telling me which toolchain you are using? I wasn't aware that Kotlin has standalone support for wasm, I would love to give it a try if it's publicly accessible

bashor, 2024-05-10:
I'm working on Kotlin/Wasm (with other folks), so feel free to reach out if you have any questions.

TianlongLiang, 2024-05-13:
Thanks, I will definitely check it out!

wenyongh, 2024-05-20:
Please try #3447.

bashor, 2024-06-04:
Hi, @wenyongh @TianlongLiang! Thanks for fixes, it really helped with attached examples. 

Previous binaries was compiled with disabled EH on Kotlin side, after turning it on I got the a similar error. 
Here is a new binary --[kotlin-wasm-wasi-example-wasm-wasi.wasm.zip](https://github.com/user-attachments/files/15570015/kotlin-wasm-wasi-example-wasm-wasi.wasm.zip)

wenyongh, 2024-06-06:
@bashor Welcome. Could you upload the wast file also? It is a little complex to analyze the wasm file without wast info, I used the tool compiled from wasm GC MVP spec proposal, but it reports: `decoding error: malformed section id`.

bashor, 2024-06-07:
@wenyongh 
Here is a wat file I got using [Binaryen's](https://github.com/WebAssembly/binaryen/) wasm-dis -- [kotlin-wasm-wasi-example-wasm-wasi.wasm.wat.zip](https://github.com/user-attachments/files/15741154/kotlin-wasm-wasi-example-wasm-wasi.wasm.wat.zip)

```shell
$ wasm-dis --enable-gc --enable-reference-types --enable-exception-handling --enable-bulk-memory kotlin-wasm-wasi-example-wasm-wasi.wasm -o  kotlin-wasm-wasi-example-wasm-wasi.wasm.wat
```

wenyongh, 2024-06-18:
@bashor I tried to debug it but since I am not familiar with the exception handling, I am not sure whether it is caused by the case itself, or by exception handling, or by the GC implementation. And the wasm file is really big and there are too many GC types, I need more time to try.

bashor, 2024-06-19:
@wenyongh Hi! 
Here is a smaller version -- [kotlin-wasm-wasi-example-wasm-wasi.wasm-prod.zip](https://github.com/user-attachments/files/15893997/kotlin-wasm-wasi-example-wasm-wasi.wasm-prod.zip). 
Hope it will be easier to inspect. 

--
Also, you can find a version optimized with Binaryen here.
[kotlin-wasm-wasi-example-wasm-wasi.wasm-prod-opt.zip](https://github.com/user-attachments/files/15893998/kotlin-wasm-wasi-example-wasm-wasi.wasm-prod-opt.zip)

With the optimized version, wami fails differently:
```
iwasm(872,0x1f55e8c00) malloc: *** error for object 0x600000000003: pointer being freed was not allocated
iwasm(872,0x1f55e8c00) malloc: *** set a breakpoint in malloc_error_break to debug
zsh: abort 
```

wenyongh, 2024-07-02:
@bashor thanks for the wat files and the new samples, there are several issues in the handling of exception handling and GC, I submitted PR #3586 to fix them and now all the files run successfully. Please try again.

bashor, 2024-07-02:
@wenyongh Nice, thanks! I'll try it and let you know.

bashor, 2024-07-02:
@wenyongh confirm, with #3586, [the simple example](https://github.com/Kotlin/kotlin-wasm-examples/tree/main/wasi-example) works. Thanks!
I found some other bugs and am going to file separate issues for them.

wenyongh, 2024-07-05:
@bashor welcome, please feel free to file issues, we will try to fix them.

bashor, 2024-07-05:
@wenyongh JFYI I've just filed #3598 and #3599

bennyhuo, 2025-06-17:
It seems that the issue is reintroduced by 928598f1ce9157360e30ebe9522d04e82266453e.

bennyhuo, 2025-06-20:
> @bennyhuo Hi, thanks for your discovery. I have created a new PR #4372 and tried running the first WASM binary from the issue — it now produces the correct result. Could you have a try, or wait for this PR to be merged into the main branch?

Awesome. I'll try it later.

bennyhuo, 2025-06-20:
> [@bennyhuo](https://github.com/bennyhuo) Hi, thanks for your discovery. I have created a new PR [#4372](https://github.com/bytecodealliance/wasm-micro-runtime/pull/4372) and tried running the first [...]

It works now. Thanks.