wasm-micro-runtime/wasm-micro-runtime #3416: exception handler intercepts c++ exceptions in the host on windows

malytomas, 2024-05-12:
embedding wamr into c++ application, after initializing, the wamr spams into console with every c++ exception thrown (in the host), even if completely unrelated to wasm.

the logging happens on https://github.com/bytecodealliance/wasm-micro-runtime/blob/c85bada2a9c85bc80f4c1a6baa7cefb662013eb4/core/iwasm/common/wasm_runtime_common.c#L386

my understanding is that this handler is used for HW bounds checks, in which case it should ignore all other types of exceptions, and only log when actual issue has happened.

TianlongLiang, 2024-05-22:
So how about LOG_ERROR for the actual wasm exception and LOG_WARNING/LOG_VERBOSE to replace the log you are pointing out? And maybe add extra info like "unrecognized exception for wasm, pass to user's host exception handler" to explain the intention of return EXCEPTION_CONTINUE_SEARCH is to search for the next handler(transfer to the user's own handler).  What do you think?

malytomas, 2024-05-22:
Hi, I have looked into the code in more detail now.

I would prefer to log a warning when inside wasm, and unknown exception code only. There is no point in logging anything if the exception is not related to the wasm, as such cases should be handled in other handlers. And no logging is necessary when the exception code is handled with wasm_set_exception.

```
diff --git a/core/iwasm/common/wasm_runtime_common.c b/core/iwasm/common/wasm_runtime_common.c
index c6e9f0b9..3dbcc76d 100644
--- a/core/iwasm/common/wasm_runtime_common.c
+++ b/core/iwasm/common/wasm_runtime_common.c
@@ -384,12 +384,16 @@ runtime_exception_handler(EXCEPTION_POINTERS *exce_info)
                 return ret;
         }
 #endif
+        else {
+            LOG_WARNING("Unhandled exception thrown: "
+                        "exception code: 0x%lx, "
+                        "exception address: %p, "
+                        "exception information: %p\n",
+                        ExceptionRecord->ExceptionCode,
+                        ExceptionRecord->ExceptionAddress, sig_addr);
+        }
     }

-    LOG_ERROR("Unhandled exception thrown:  exception code: 0x%lx, "
-              "exception address: %p, exception information: %p\n",
-              ExceptionRecord->ExceptionCode, ExceptionRecord->ExceptionAddress,
-              sig_addr);
     return EXCEPTION_CONTINUE_SEARCH;
 }
 #endif /* end of BH_PLATFORM_WINDOWS */
```

Furthermore, It would be nice to have similar logging on linux, which currently does not log anything. However, I dont know if it is allowed to do so inside signal handler?

TianlongLiang, 2024-05-31:
I think the current method in Linux is fine. If you look at the code logic here: https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/shared/platform/common/posix/posix_thread.c#L610-L643
If the signal handler of the runtime cannot handle it, the runtime will first check whether there are other signal handlers registered previously. If there are, it will be transferred to it. If not, it will finally log it.

malytomas, 2024-05-31:
awesome. thanks ;)

closed 2024-05-31 by malytomas