wasm-micro-runtime/wasm-micro-runtime #3432: Can AOT file built on windows run directly on other platforms?

kamylee, 2024-05-14:
Can the AOT file built on windows run directly on other platforms? like linux ubuntu22.

TianlongLiang, 2024-05-17:
I think so, I have built an AOT file on Ubuntu20 and it can run on my Windows

kamylee, 2024-05-21:
> I think so, I have built an AOT file on Ubuntu20 and it can run on my Windows

Thank you! I'll try again.

kamylee, 2024-05-22:
1、I created an AOT file on Windows, and when I run it on Ubuntu 22.04, it shows the error: wasm_runtime_load: AOT module load failed: invalid data relocation section name.
2、I created an AOT file on Ubuntu22.04 and run it on windows. error: wasm_runtime_load:AOT module load failed: relocation truncated to fit IMAGE_REL_AMD64_ADDR32 failed. Try using wamrc with --size-level=1 option.

But it runs well in wasm mode.

TianlongLiang, 2024-05-22:
My bad, I forgot to mention how to do cross-compilation for AOT. This error occurs because the relocation type in your two AOT files does not conform to the native compatible format. What you are trying to achieve is like cross-compilation, it will require some certain cross-compiling toolchain to be used.

> PS: The following flags are just demoing my points, you should check what's right for your situation

Here is what you would do cross-compile normally:

```bash
# This package provides the necessary headers and libraries for Windows, and tools for cross-compilation.
sudo apt install mingw-w64
clang --target=x86_64-w64-mingw32 --sysroot=/usr/x86_64-w64-mingw32 -o myprogram.exe myprogram.c
```

And what you need to do in aot compilation is similar, something like:
```
WAMRC_LLC_COMPILER=clang WAMRC_LLC_FLAGS="--target=x86_64-w64-mingw32 --sysroot=/usr/x86_64-w64-mingw32" ./wamrc -o test.aot test.wasm 
```

Here is what that environment variable means: https://github.com/bytecodealliance/wasm-micro-runtime/blob/591a20b91741c24cdb9c031d3c5eec1d80bd72dc/doc/build_wasm_app.md#aot-compilation-with-3rd-party-toolchains

BTW, Generally, I only do cross-compilation when I absolutely have to, otherwise I think it's just easier to set up development environment on the other platform

TianlongLiang, 2024-05-22:
Can you try `wamrc --target=gnu` on windows to see whether it will work?