wasm-micro-runtime/wasm-micro-runtime #3442: iwasm with aot file fail with SIGSEGV

peter-jerry-ye, 2024-05-16:
### Subject of the issue

`iwasm` with an aot fails with `unhandled SIGSEGV`. The original `wasm` works fine.

### Test case

WAT file: [test.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/15330646/test.txt)

A Moonbit version of [n-body simulation](https://github.com/web-devkits/Wasmnizer-ts/blob/main/tests/benchmark/nbody_class.ts)

### Your environment

* Host OS: Mac M1 Sonoma 14.4.1
* WAMR version, platform, cpu architecture, running mode: 7949df96f4d3a32170fe540ffc581f39af4d2c07, iwasm built with GC support

### Steps to reproduce

```bash
wasm-tools parse test.wat -o test.wasm
wamrc --size-level=0 -o test.aot --enable-gc test.wasm # size-level=0 due to Mac M1
iwasm test.aot
```

### Expected behavior

Print result (which I assume is `0x0:i32`)

### Actual behavior

`unhandled SIGSEGV, si_addr: 0x67ffff79`

TianlongLiang, 2024-05-20:
Hi, I run it on my x86 ubuntu and didn't reproduce the error, I think it's platform specific bug. Can you try run iwasm with -v=5 to print log so that it may help locate the bug?

peter-jerry-ye, 2024-05-20:
> Hi, I run it on my x86 ubuntu and didn't reproduce the error, I think it's platform specific bug. Can you try run iwasm with -v=5 to print log so that it may help locate the bug?

Sure, here's the log:
```
[23:00:22:227 - 1F8183AC0]: type 0: struct, field count: 7, ref type map count: 0
[23:00:22:254 - 1F8183AC0]:                 field: 0, flags: 1, type: 124
[23:00:22:257 - 1F8183AC0]:                 field: 1, flags: 1, type: 124
[23:00:22:259 - 1F8183AC0]:                 field: 2, flags: 1, type: 124
[23:00:22:262 - 1F8183AC0]:                 field: 3, flags: 1, type: 124
[23:00:22:264 - 1F8183AC0]:                 field: 4, flags: 1, type: 124
[23:00:22:266 - 1F8183AC0]:                 field: 5, flags: 1, type: 124
[23:00:22:268 - 1F8183AC0]:                 field: 6, flags: 0, type: 124
[23:00:22:271 - 1F8183AC0]: type 1: array
[23:00:22:274 - 1F8183AC0]: type 2: func, param count: 0, result count: 1, ref type map count: 1
[23:00:22:277 - 1F8183AC0]: type 3: func, param count: 0, result count: 1, ref type map count: 0
[23:00:22:280 - 1F8183AC0]: type 4: func, param count: 1, result count: 1, ref type map count: 0
[23:00:22:282 - 1F8183AC0]: type 5: func, param count: 1, result count: 1, ref type map count: 0
[23:00:22:838 - 1F8183AC0]: type 6: func, param count: 0, result count: 1, ref type map count: 0
[23:00:22:846 - 1F8183AC0]: type 7: func, param count: 0, result count: 0, ref type map count: 0
[23:00:22:923 - 1F8183AC0]: Load module success.

[23:00:22:981 - 1F8183AC0]: Memory instantiate:
[23:00:22:984 - 1F8183AC0]:   page bytes: 65536, init pages: 1, max pages: 1
[23:00:22:987 - 1F8183AC0]:   data offset: 0, stack size: 0
[23:00:22:989 - 1F8183AC0]:   heap offset: 65536, heap size: 0

unhandled SIGSEGV, si_addr: 0x60003f79
zsh: abort      ~/Documents/wasm-micro-runtime/product-mini/platforms/darwin/build/iwasm -v=5
```

TianlongLiang, 2024-05-20:
So if you run wasm file, will it raise the same error?

peter-jerry-ye, 2024-05-20:
> So if you run wasm file, will it raise the same error?

No, only the aot file fails