wasm-micro-runtime/wasm-micro-runtime #3466: How to croos compile to riscv64

HoyiTT, 2024-05-23:
Hi Everyone.
I want to build iwasm for riscv64 on x86_64 ubuntu(docker)
Bur it's doesn't works.
I referred to the following link.  https://github.com/bytecodealliance/wasm-micro-runtime/issues/1830

I installed RISC-V GNU Compiler Toolchain  and  export PATH=/opt/riscv/bin:$PATH, and then added lines to CMakeLists.txt

+SET(CMAKE_C_COMPILER riscv64-unknown-linux-gnu-gcc)
+SET(CMAKE_AR riscv64-unknown-linux-gnu-ar)
+SET(CMAKE_RANLIB riscv64-unknown-linux-gnu-ranlib)
+SET(WAMR_BUILD_TARGET "RISCV64")

and i tried

cmake ..
make

But it's doesn't build..

/root/wasm-micro-runtime-main/core/iwasm/common/arch/invokeNative_riscv.S:144:9: error: invalid instruction mnemonic 'mv'
        mv sp, fp
        ^~
/root/wasm-micro-runtime-main/core/iwasm/common/arch/invokeNative_riscv.S:145:22: error: expected register here
        ld fp, 0 * 8(sp)
                     ^
/root/wasm-micro-runtime-main/core/iwasm/common/arch/invokeNative_riscv.S:146:22: error: expected register here
        ld ra, 1 * 8(sp)
                     ^
/root/wasm-micro-runtime-main/core/iwasm/common/arch/invokeNative_riscv.S:147:9: error: invalid instruction mnemonic 'addi'
        addi sp, sp, 2 * 8
        ^~~~
/root/wasm-micro-runtime-main/core/iwasm/common/arch/invokeNative_riscv.S:148:9: error: invalid instruction mnemonic 'jr'
        jr ra
        ^~
make[2]: *** [CMakeFiles/vmlib.dir/build.make:621: CMakeFiles/vmlib.dir/root/wasm-micro-runtime-main/core/iwasm/common/arch/invokeNative_riscv.S.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/vmlib.dir/all] Error 2
make: *** [Makefile:136: all] Error 2`

Plesea anyone helpme!

wjdrud2532, 2024-05-24:
Hello

If you applied the version, the path, and all that

**Delete CmakeCache.txt** in the cmake path and proceed with the cmake again

thank you

HoyiTT, 2024-05-24:
OMG

I installed riscv64-glibc-ubuntu-20.04-nightly-2022.12.17-nightly.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2022.12.17/riscv64-glibc-ubuntu-20.04-nightly-2022.12.17-nightly.tar.gz

and Delete CmakeCahec.tx and then 

```
cmake ..
make
```

it work's!

Thank you!!!

closed 2024-05-24 by HoyiTT