wasm-micro-runtime/wasm-micro-runtime #3468: A bug in validation

erxiaozhou, 2024-05-23:
### Subject of the issue
There is a bug in the validation phase of WAMR, and the bug can be triggered by the following test case. 
The test case is invalid since it contains an illegally encoded block type (at offset 0x32e2). However, the test case can pass the validation of multiple runtimes, including classic interpreter, fast interpreter, JIT mode, fast JIT mode.

### Test case

[i64.add_2277BSBBJJJJJPSSSSSSSSSBBB_331_17164376207829096.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/15421510/i64.add_2277BSBBJJJJJPSSSSSSSSSBBB_331_17164376207829096.zip)

### Your environment

* Host OS: Ubuntu 20.04
* Commit Id: f378edc408382b7bcf517b463d4a54948739075b

### Steps to reproduce
1. build WAMR 
2. execute the command. E.g., 
```
iwasm --heap-size=0 -f "to_test" <test_case>  
```
### Expected behavior
An exception indicating "invalid function type"
### Actual behavior
An exception
```
0x0:i64
```

wenyongh, 2024-05-30:
@erxiaozhou thanks for spotting the issue! I submitted PR #3482 to fix it, please try again.

wenyongh, 2024-06-03:
Close the issue as it was fixed.

closed 2024-06-03 by wenyongh