wasm-micro-runtime/wasm-micro-runtime #3475: Calling "step-over" on the last line of the function failed

WNsACE, 2024-05-27:
Here is the assembly of test code
[image]

When the breakpoint is triggered at "fun [79] return", as shown in the following figure:
[image]

At this point, the runtime is in the following code:
[image]

When the user calls "step-over", the following code will be triggered
[image]

But the value of "step_count" is 1, so the pause signal will not be triggered
[image]

At the same time, the value of "step_count" is 1, so the pause signal will not be triggered. However, "return_func" restores the stack, so only the next section of code can be executed. If the next line of code is a calling function, it will enter the function and cause debugging information errors.

**May I ask why "step_count" needs to be equal to 2? And how to resolve the appeal issue?**

TianlongLiang, 2024-06-17:
I will take a look at this issue, and keep you updated

TianlongLiang, 2024-07-01:
You are right, this line should be a typo, it should be 1, I submitted PR #3583 for this issue. Thanks for spotting it and reporting it!