wasm-micro-runtime/wasm-micro-runtime #3504: Exception: out of bounds memory access

kamylee, 2024-06-05:
When i run following code in wasm.i receive an error message:Exception: out of bounds memory access
`
cJSON* jRoot = cJSON_CreateObject();
for (int i = 0; i < 47; i++)//Greater than 46 will prompt: out of bounds memory access
{
	string strId = "id"+ std::to_string(i);
	cJSON_AddStringToObject(jRoot, strId.c_str(), std::to_string(i).c_str());
}
`
But run this code in native is ok。
**running platform**:windows10

this is my set:
[image]

`
static char global_heap_buf[2*1024* 1024];
...
init_args.mem_alloc_option.pool.heap_buf = global_heap_buf;
init_args.mem_alloc_option.pool.heap_size = sizeof(global_heap_buf);
`
Increased the memory, but the problem still exists. how to solve it?

TianlongLiang, 2024-06-05:
Hi, could you try increate `-zstack-size` and `--max-memory`?

kamylee, 2024-06-05:
Yes. I hava already tried to increasing the momery for -zstack-size and --max-memory by 10 times, But the broblem has not been resolved.

TianlongLiang, 2024-06-06:
I think it may be helpful to take a look at: https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/debug-tools You can locate which line of the source code to throw an exception

kamylee, 2024-08-07:
Thanks! I replaced the cJSON with the jsoncpp and it worked fine!

closed 2024-08-07 by kamylee