wasm-micro-runtime/wasm-micro-runtime #3505: How to debug wasm on the Windows platform

w4454962, 2024-06-05:
When installing the WARM IDE in vscode, it prompts that LLDB does not support the Windows platform

TianlongLiang, 2024-06-06:
Sadly for now we don't have much support for Windows lldb, therefore you can't use the debug feature in Windows WAMR IDE. You may try other features to help debug like https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#enable-dump-call-stack-feature or https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/debug-tools

w4454962, 2024-06-06:
thanks, I hope to support debugging in a Win32 environment in the future, even in interpreted mode.

vkensou, 2024-06-24:
希望对windows支持可以更好点。现在的release里也没有windows平台的包可以下载。

iKlask, 2024-07-03:
> Sadly for now we don't have much support for Windows lldb, therefore you can't use the debug feature in Windows WAMR IDE. You may try other features to help debug like [...]

Is windows not supported at all even if you build LLDB yourself and [use the following instructions](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/test-tools/wamr-ide/VSCode-Extension/resource/debug)? I built lldb.exe and lldb-vscode.exe for windows and I would like to try debugging on windows.

I did notice the extension doesn't look for the correct path in [`isLLDBInstalled`](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/test-tools/wamr-ide/VSCode-Extension/src/utilities/lldbUtilities.ts#L54) when on windows. It uses `os.platform()` which evaluates to "win32" but the instructions want to use the path with "windows" as the folder name. With some adjustments my WAMRIDE extension detects lldb on windows now, however it seems to fail when trying to launch the debug server? after the extension calls `boot_debugger_server.bat` I get the following:

```
[14:21:59:921 - 7FDD7CF18000]: warning: allocate 544768 bytes memory failed
[14:21:59:948 - 7FDD7CF18000]: WASM Debug Engine warning: failed to allocate linear memory for execution.
Will not be able to evaluate expressions during debugging
[14:21:59:090 - 7FDD7CF0E640]: control thread of debug object 0x555ee216efe0 start
[14:21:59:172 - 7FDD7CF0E640]: Debug server listening on 0.0.0.0:1234
[14:21:59:221 - 7FDD7CF18000]: warning: allocate 28 bytes memory failed
Exception: allocate memory failed
[14:21:59:314 - 7FDD7CF0E640]: wasm gdb server error: socket accept failed
[14:21:59:325 - 7FDD7CF0E640]: Failed while accepting debugger connection
```
and lldb afterwards prints

```
Running attachCommands:
(lldb) process connect -p wasm connect://127.0.0.1:1234
error: Failed to connect port
```

im not sure if its failing because of those warnings about memory allocation or a different reason

TianlongLiang, 2024-07-04:
I haven't tried this on Windows, but you can try add --heap-size=1024000 option to iwasm and it should resolve memory allocation issue

iKlask, 2024-07-12:
I got debugging working on the windows platform. You do have to build LLDB yourself with the supplied patch, but there's a few other issues I had to clear up to get it working:

1. I had to [build LLDB myself](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/source_debugging_interpreter.md#debugging-with-interpreter) and [place the output binaries/libs in the extension folder manually](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/test-tools/wamr-ide/VSCode-Extension/resource/debug)
2. When building lldb, I had to make sure libxml2 was installed on windows. CMAKE for LLVM also expects v2.8 of libxml2 and will not work with newer versions. I manually disabled this in the .cmake files of LLVM to allow my latest installation of libxml2 to work (2.11.8). Without libxml2, lldb will still build and execute normally... but it will never attempt to read debug symbols from the debug server
3. I had to modify the code in [isLLDBInstalled](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/test-tools/wamr-ide/VSCode-Extension/src/utilities/lldbUtilities.ts#L54) so it wouldn't use `os.platform()` when picking the path to the lldb bin folder. This way it should look in the `resource\debug\windows\` folder instead of `resource\debug\win32\`
4. I had to disable the use of docker for the debug server (iwasm). The iwasm docker image on windows would never communicate over localhost to the lldb instance. I dont know if this is an issue with my own setup or normal behavior for docker on windows, because this worked in linux. I had to edit the `boot_debugger_server.bat` file to replace the docker line with a call to iwasm.exe that I had manually built:
`path/to/iwasm/iwasm.exe --heap-size=%3 -g=127.0.0.1:1234 %cd%\build\%1.wasm`