wasm-micro-runtime/wasm-micro-runtime #3510: Failure to execute AOT code compiled to 32bit --target=i386

iKlask, 2024-06-06:
I've been using WAMR with intentions of embedding it within a 32bit app on windows. After pulling 2.0.0 I've noticed my AOT execution gets stuck in iwasm. Specifically inside the `invoke_native` call. If I compile for x86_64 AOT it works fine in x64 iwasm.

I've attached a simple wasm example (Fibonacci function). Both the source .wasm file and the compiled .aot from wamrc on 2.0.0 for --target=i386
[wasm.zip](https://github.com/user-attachments/files/15677059/wasm.zip)

wenyongh, 2024-06-11:
@iKlask 32bit may take longer time to run fib with the same argument, and it may take very long time to run fib with large input argument, had you tested the 32bit AOT file with a small argument? e.g. `iwasm -f fib fib_i386.aot 10`?

iKlask, 2024-06-11:
yes I tried with many different arguments. My usual for testing was 10, 25, 30, and 40. This did work at one point since I had a benchmark setup. Hadn't tried AOT on 32bit in a while and now it locks up. I don't know if its an issue with whats generated in WAMRC, or if its an issue with the runtime.

iKlask, 2024-06-14:
I reduced the threashold multiple times and it never logs. I even went down to 1 and nothing. Here is an example of fib(5) with a threshold of 1 on both x86 and x64 iwasm/AOT. The x86 version hangs before it can even log. the x64 version works fine

```
iwasm-x86.exe -v=5 -f fib fibLog.aot 5
[19:59:55:568 - D03CC8]: Load module success.

[19:59:55:569 - D03CC8]: Memory instantiate:
[19:59:55:569 - D03CC8]:   page bytes: 16384, init pages: 1, max pages: 1
[19:59:55:569 - D03CC8]:   data offset: 0, stack size: 0
[19:59:55:569 - D03CC8]:   heap offset: 0, heap size: 16384

^C
iwasm-x64.exe -v=5 -f fib fibLogX64.aot 5
[20:00:01:327 - 7FF6F4F14750]: Load module success.

[20:00:01:327 - 7FF6F4F14750]: Memory instantiate:
[20:00:01:327 - 7FF6F4F14750]:   page bytes: 16384, init pages: 1, max pages: 1
[20:00:01:327 - 7FF6F4F14750]:   data offset: 0, stack size: 0
[20:00:01:327 - 7FF6F4F14750]:   heap offset: 0, heap size: 16384

in specttest.print_i32(2)
in specttest.print_i32(1)
in specttest.print_i32(3)
in specttest.print_i32(2)
in specttest.print_i32(2)
in specttest.print_i32(1)
in specttest.print_i32(2)
in specttest.print_i32(1)
in specttest.print_i32(3)
in specttest.print_i32(2)
in specttest.print_i32(4)
in specttest.print_i32(3)
in specttest.print_i32(5)
in specttest.print_i32(5)
0x5:i32
```

iKlask, 2024-06-24:
Yes, I do use -DWAMR_BUILD_TARGET=X86_32 for my 32bit builds of iwasm

iKlask, 2024-06-25:
okay that patch does seem to work for 32bit AOT

iKlask, 2025-01-29:
I'm not sure what has changed but I haven't been able to get win32 AOT working on any 2.0.0+ update. in iwasm when invoke_native is called it seems to jump to incorrect or bad instructions. Usually it instantly returns.

I've tried both on the releases for 2.0.0, 2.1.0, and 2.2.0. My wamrc command to compile for 32 bit windows has been very simple:
`wamrc --target=i386 -o output.aot input.wasm`

I have tried both with and without this `need_precheck = false;` and it doesnt seem to change how my iwasm executes the aot code.