wasm-micro-runtime/wasm-micro-runtime #3514: Heap Buffer Overflow inside "wasm_loader_prepare_bytecode" Function

mobsceneZ, 2024-06-07:
### Subject of the issue

Running the CLI iwasm with the given testcase results in segmentation fault, which is caused by heap buffer overflow.

### Test case

[iwasm-poc-05.zip](https://github.com/user-attachments/files/15740484/iwasm-poc-05.zip)

### Your environment

```shell
OS               : Linux 5.15.146.1-microsoft-standard-WSL2 #1 SMP Thu Jan 11 04:09:03 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
Commit           : 028f43bc18494866c44666e54e9c5a2cd84152f5
Version          : 2.1.0
Clang Verison    : 13.0.0
Affected Tool    : iwasm
Enabled Features : Bulk Memory, Reference Types
```
### Steps to reproduce

```shell
Build            : cd product-mini/platforms/linux/ && mkdir -p build && cd build && export CC="clang" CXX="clang++" CFLAGS="-fsanitize=address -g" CXXFLAGS="-fsanitize=address -g" && cmake -DWAMR_BUILD_BULK_MEMORY=1 -DWAMR_BUILD_REF_TYPES=1 .. && make -j
Command          : iwasm -f main iwasm-poc-05
```

### Expected behavior

The program should exit gracefully with possibly some error information.

### Actual behavior

Here is the stack trace provided by AddressSanitizer:
```shell
=================================================================
==23895==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6060000001fe at pc 0x55ea15ed1c32 bp 0x7fff717e3730 sp 0x7fff717e3728
READ of size 2 at 0x6060000001fe thread T0
    #0 0x55ea15ed1c31 in wasm_loader_prepare_bytecode /home/lain/wamr/core/iwasm/interpreter/wasm_loader.c:12145:30
    #1 0x55ea15ea9c10 in load_from_sections /home/lain/wamr/core/iwasm/interpreter/wasm_loader.c:6130:14
    #2 0x55ea15ead2c5 in load /home/lain/wamr/core/iwasm/interpreter/wasm_loader.c:6536:13
    #3 0x55ea15ead2c5 in wasm_loader_load /home/lain/wamr/core/iwasm/interpreter/wasm_loader.c:6713:10
    #4 0x55ea15e42f57 in wasm_runtime_load_ex /home/lain/wamr/core/iwasm/common/wasm_runtime_common.c:1353:33
    #5 0x55ea15e432b0 in wasm_runtime_load /home/lain/wamr/core/iwasm/common/wasm_runtime_common.c:1398:12
    #6 0x55ea15e3e248 in main /home/lain/wamr/product-mini/platforms/linux/../posix/main.c:924:25
    #7 0x7fe5d8d45082 in __libc_start_main /build/glibc-LcI20x/glibc-2.31/csu/../csu/libc-start.c:308:16
    #8 0x55ea15d6e6bd in _start (/home/lain/wamr/product-mini/platforms/linux/build/iwasm+0x396bd)

0x6060000001fe is located 2 bytes before 64-byte region [0x606000000200,0x606000000240)
allocated by thread T0 here:
    #0 0x55ea15e05e1e in __interceptor_malloc /home/build-user/llvm-project/compiler-rt/lib/asan/asan_malloc_linux.cpp:69:3
    #1 0x55ea15eae11f in loader_malloc /home/lain/wamr/core/iwasm/interpreter/wasm_loader.c:384:20
    #2 0x55ea15eae11f in wasm_loader_ctx_init /home/lain/wamr/core/iwasm/interpreter/wasm_loader.c:8215:15
    #3 0x55ea15eae11f in wasm_loader_prepare_bytecode /home/lain/wamr/core/iwasm/interpreter/wasm_loader.c:10954:24
    #4 0x55ea15ea9c10 in load_from_sections /home/lain/wamr/core/iwasm/interpreter/wasm_loader.c:6130:14
    #5 0x55ea15ead2c5 in load /home/lain/wamr/core/iwasm/interpreter/wasm_loader.c:6536:13
    #6 0x55ea15ead2c5 in wasm_loader_load /home/lain/wamr/core/iwasm/interpreter/wasm_loader.c:6713:10
    #7 0x55ea15e42f57 in wasm_runtime_load_ex /home/lain/wamr/core/iwasm/common/wasm_runtime_common.c:1353:33
    #8 0x55ea15e3e248 in main /home/lain/wamr/product-mini/platforms/linux/../posix/main.c:924:25
    #9 0x7fe5d8d45082 in __libc_start_main /build/glibc-LcI20x/glibc-2.31/csu/../csu/libc-start.c:308:16

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/lain/wamr/core/iwasm/interpreter/wasm_loader.c:12145:30 in wasm_loader_prepare_bytecode
Shadow bytes around the buggy address:
  0x605fffffff00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x605fffffff80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x606000000000: fa fa fa fa 00 00 00 00 00 00 00 fa fa fa fa fa
  0x606000000080: 00 00 00 00 00 00 00 fa fa fa fa fa 00 00 00 00
  0x606000000100: 00 00 00 fa fa fa fa fa 00 00 00 00 00 00 00 fa
=>0x606000000180: fa fa fa fa 00 00 00 00 00 00 00 fa fa fa fa[fa]
  0x606000000200: 00 00 00 00 00 00 00 00 fa fa fa fa fa fa fa fa
  0x606000000280: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x606000000300: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x606000000380: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x606000000400: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==23895==ABORTING
```

wenyongh, 2024-06-11:
@mobsceneZ Thanks for spotting the issue! I submitted PR #3516 to fix it, could you please help have a look and try again?

mobsceneZ, 2024-06-11:
@wenyongh Great! Seems like both issue #3513 and issue #3514 are fixed by PR #3516, thanks for your effort!

wenyongh, 2024-06-13:
@mobsceneZ Welcome. Will merge the PR after it is reviewed.