wasm-micro-runtime/wasm-micro-runtime #3521: Issue with running Pico on wamr through Zephyr

TAOFOR4, 2024-06-12:
Hi,

I am trying to run wasm code on Pico through wamr and Zephyr by using command "west build -b rpi_pico -p always -- -DWAMR_BUILD_TARGET=THUMB"

It works. However after flashing the .uf2 file into Pico. I didn't see the output of Pico which is usually shown as "/dev/ttyACM0" in my linux system. I though it might be USB port setting issue but I am not sure how to set the prj.conf file(or maybe somewhere else). Here is my prj.conf. I added some lines based on other zephyr Pico projects. But it doesn't work. I am unfamiliar with this area so come here for help.

```

# Enable threads, necessary for WAMR's execution
CONFIG_MULTITHREADING=y
CONFIG_NUM_PREEMPT_PRIORITIES=15

# Logging for debugging
CONFIG_LOG=y

# Enable MPU for memory protection (might be needed for WAMR security features)
CONFIG_ARM_MPU=y

CONFIG_ARM_MMU=n

CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PID=0x0004
CONFIG_USB_DEVICE_INITIALIZE_AT_BOOT=n

CONFIG_GPIO=y
CONFIG_FLASH=y
CONFIG_SERIAL=y
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y
CONFIG_UART_LINE_CTRL=y

CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_HEAP_MEM_POOL_SIZE=1024
```

Thx

TAOFOR4, 2024-06-12:
I also found same issue for nRF5340. "screen /dev/ttyACM1 115200" can't check the output

deadprogram, 2024-08-23:
I have the same issue. Can build the UF2 file, but flashing it to the Pico no serial port appears. I suspect that it probably cannot run on the Pico due to not having enough memory? Anyone else have any thought on this?