wasm-micro-runtime/wasm-micro-runtime #3524: OSS-Fuzz issue 69577

oss-fuzz-robot, 2024-06-12:
OSS-Fuzz has found a bug in this project. Please see https://oss-fuzz.com/testcase?key=6265900542525440 for details and reproducers.

This issue is mirrored from https://bugs.chromium.org/p/oss-fuzz/issues/detail?id=69577 and will auto-close if the status changes there.

If you have trouble accessing this report, please file an issue at https://github.com/google/oss-fuzz/issues/new.

yamt, 2024-06-13:
> OSS-Fuzz has found a bug in this project. Please see https://oss-fuzz.com/testcase?key=6265900542525440 for details and reproducers.
> 
> This issue is mirrored from https://bugs.chromium.org/p/oss-fuzz/issues/detail?id=69577 and will auto-close if the status changes there.

i don't have permissions on the above urls. is it expected?

@lum1n0us any idea?

lum1n0us, 2024-06-13:
#3526

oss-fuzz-robot, 2024-06-14:
OSS-Fuzz has closed this bug. Please see https://bugs.chromium.org/p/oss-fuzz/issues/detail?id=69577 for details.

closed 2024-06-14 by oss-fuzz-robot