wasm-micro-runtime/wasm-micro-runtime #3531: Fail allocate but print success?

Userzxcvbvnm, 2024-06-15:
### Subject of the issue
In this situation, wamr fails to allocate space which is expected ,however, wamr prints success.

### Test case
The test case is:
```

#include <stdio.h>
#include <stdlib.h>
#include <fcntl.h>
#include <unistd.h>
#include <sys/types.h>
#include <sys/stat.h>

void print_file_size(int fd){
    struct stat st;
    if (fstat(fd, &st) == -1) {
        printf("Get file size failed.\n");
    } else {
        printf("Get file size: %ld bytes.\n", st.st_size);
    }
}

int get_fd(const char *filename, int flags) {
    int fd = open(filename, flags);
    
    if (fd == -1) {
        printf("Get file descriptor of file %s failed!\n", filename);
        return -1;
    } else {
        printf("Get file descriptor of file %s succeed!\n", filename);
        return fd;
    }
}

void closebyfd(int fd) {
    if (close(fd) == -1) {
        printf("Close the file by descriptor failed!\n");
    }
}

void fd_allocate_00085_gxX49(int fd) {
    printf("Enter function fd_allocate_00085_gxX49\n");
    
    int result = posix_fallocate(fd, 1, 7);

    if (result == 0) {
        printf("Space allocation in file successful.\n");
    } else {
        printf("Error allocating space in file.\n");
    }
}

int main() {
    int fd = get_fd("subdir_2/subfile_1", O_WRONLY | O_APPEND);
    if (fd == -1) {
        return -1; // Return from main if get_fd failed
    }

    print_file_size(fd);
    fd_allocate_00085_gxX49(fd);
    print_file_size(fd);

    closebyfd(fd);

    return 0;
}

```

### Your environment
Ubuntu 20.04
x86_64
WAMR 1.3.2

### Steps to reproduce

(1)compile to wasm:./wasi-sdk-21.0/bin/clang --target=wasm32-unkown-wasi --sysroot=./wasi-sdk-21.0/share/wasi-sysroot test.c -o test.wasm

(2)Running wasm:
(Before run the Wasm file, file subdir_2/subfile_1 exists, and file size is 26.)
iwasm --dir=. test.wasm

### Expected behavior
prints:
```
Get file descriptor of file subdir_2/subfile_1 succeed!
Get file size: 26 bytes.
Enter function fd_allocate_00085_gxX49
Error allocating space in file.
Get file size: 26 bytes.
```
This is what wasmtime does.

### Actual behavior
WAMR prints:
```
Get file descriptor of file subdir_2/subfile_1 succeed!
Get file size: 26 bytes.
Enter function fd_allocate_00085_gxX49
Space allocation in file successful.
Get file size: 26 bytes.
```

### Extra Info

Anything else you'd like to add?

lum1n0us, 2024-06-25:
IIUC, you have two questions:
1. why wasmtime failed but wamr pass
2. why wamr says "Space allocation in file successful." but the before and after file size are same.

I am just going to answer the second one 😄 . 

According to [manual](https://man7.org/linux/man-pages/man3/posix_fallocate.3.html), 
> The function posix_fallocate() **ensures that disk space** is allocated for the file referred to by the file descriptor fd for the bytes.  ... If the size of the file is less than offset+len, then the file is increased to this size; otherwise the file size is left unchanged.

It is not going to increase the file size if there is enough space. In this case, the file size is 26, and `offset+len` is 7. So, the *file size is left unchanged*.

And for sure, you'll see a file size change if use a bigger `len`, like
`int result = posix_fallocate(fd, 1, 70);`

yamt, 2024-06-27:
> why wasmtime failed but wamr pass

because wasmtime has deprecated fd_allocate while ago, it always fails.

this doesn't seem like a bug in wamr.