wasm-micro-runtime/wasm-micro-runtime #3532: File allocation bug?

Userzxcvbvnm, 2024-06-15:
### Subject of the issue
Allocate the file size from a value which is larger the file size, WAMR fails to allocate. I'm not sure whether this is a bug.

### Test case
```

#include <stdio.h>
#include <stdlib.h>
#include <fcntl.h>
#include <unistd.h>
#include <sys/types.h>
#include <sys/stat.h>

void print_file_size(int fd){
    struct stat st;
    if (fstat(fd, &st) == -1) {
        printf("Get file size failed.\n");
    } else {
        printf("Get file size: %ld bytes.\n", st.st_size);
    }
}

int get_fd(const char *filename, int flags) {
    int fd = open(filename, flags);
    
    if (fd == -1) {
        printf("Get file descriptor of file %s failed!\n", filename);
        return -1;
    } else {
        printf("Get file descriptor of file %s succeed!\n", filename);
        return fd;
    }
}

void closebyfd(int fd) {
    if (close(fd) == -1) {
        printf("Close the file by descriptor failed!\n");
    }
}

void fd_allocate_00072_gxX49(int fd) {
    printf("Enter function fd_allocate_00072_gxX49\n");
    
    int result = posix_fallocate(fd, 1, 0);

    if (result == 0) {
        printf("Space allocation in file successful.\n");
    } else {
        printf("Error allocating space in file.\n");
    }
}

int main() {
    int fd = get_fd("subdir_1/subfile_2", O_WRONLY | O_TRUNC);
    if (fd == -1) {
        return -1; // Return from main if get_fd failed
    }

    print_file_size(fd);
    fd_allocate_00072_gxX49(fd);
    print_file_size(fd);

    closebyfd(fd);

    return 0;
}
```

### Your environment
Ubuntu 20.04
x86_64
WAMR 1.3.2

### Steps to reproduce

Steps to reproduce:
(1)compile to wasm:./wasi-sdk-21.0/bin/clang --target=wasm32-unkown-wasi --sysroot=./wasi-sdk-21.0/share/wasi-sysroot test.c -o test.wasm

(2)Running wasm:
(Before run the Wasm file, subdir_1/subfile_2 exists.)
iwasm --dir=. test.wasm

### Expected behavior

```
Get file size: 0 bytes.
Enter function fd_allocate_00072_gxX49
Space allocation in file successful.
Get file size: 1 bytes.
```

### Actual behavior
```
Get file descriptor of file subdir_1/subfile_2 succeed!
Get file size: 0 bytes.
Enter function fd_allocate_00072_gxX49
Error allocating space in file.
Get file size: 0 bytes.
```

lum1n0us, 2024-06-25:
Might want to change
` int result = posix_fallocate(fd, 1, 0);`
->
` int result = posix_fallocate(fd, 1, 70);`

> EINVAL offset was less than 0, or **len was less than or equal to 0**, or the underlying filesystem does not support the operation.

yamt, 2024-06-27:
i agree this is not a bug.