wasm-micro-runtime/wasm-micro-runtime #3544: Wasm shared heap

WenLY1, 2024-06-18:
- [X] add WAMR_BUILD_SHARED_HEAP and WASM_ENABLE_SHARED_HEAP
- [ ] add wamrc --enable-shared-heap
- [ ] consider the behavior of import memory in multi-module feature
- [ ] adapt address conversion/validation
      aot code boundary check
      wasm_runtime_invoke_native
      wasm_runtime_invoke_native_raw
     ~~wasm_runtime_validate_app_addr~~
     ~~wasm_runtime_addr_app_to_native~~
     ~~wasm_runtime_addr_native_to_app~~
     ~~wasm_runtime_validate_native_addr~~
- [ ] interpreter/fast-jit do boundary check
     ~~classic interp mode~~
     fast interp mode
     jit

- [X] Support setting shared heap’s size and shared heap’s alloc options in RuntimeInitArgs for wasm_runtime_full_init
- [ ] Add check in load module, default memory’s max_memory_size should be no larger than 4G-shared_heap_size, if not, reduce it to 4G-shared_heap_size
- [X] Add API wasm_runtime_shared_malloc/wasm_runtime_shared_free
- [X] App allocates memory from shared heap with API shared_malloc/shared_free
      example
```
            #include <stdio.h>  
            #include <stdlib.h>  
            extern void *shared_malloc(int size);  
            extern void shared_free(void *ptr);  
            int main()  
            {  
	            int *pa = (int *)malloc(4);  
	            *pa = 4;  
	            printf("pa value is %d\n", *pa);  
	            free(pa);  
	            int *ps = (int *)shared_malloc(4);  
	            *ps = 5;  
	            printf("ps value is %d\n", *ps);  
	            shared_free(ps);  
            } 
```

WenLY1, 2024-06-18:
If wasm shared heap is enable, should the boundary check keep enabled ? In the following code, if bounds checks is disabled, should we return the offset directly or judge the addr is in shared heap?
In function wasm_runtime_addr_native_to_app
```
    if (bounds_checks) {
        if (memory_inst->memory_data <= addr && addr < memory_inst->memory_data_end) {
            ret = (uint64)(addr - memory_inst->memory_data);
            SHARED_MEMORY_UNLOCK(memory_inst);
            return ret;
        }
    }
    /* If bounds checks is disabled, return the offset directly */
    else if (addr != NULL) {
        ret = (uint64)(addr - memory_inst->memory_data);
        SHARED_MEMORY_UNLOCK(memory_inst);
        return ret;
    }
```

lum1n0us, 2024-06-18:
Would u mind giving us, the community, a quick introduction about:
- What's *shared heap*? 
- What kind of questions does *shared heap* try to answer?
- Is there any other options? And why we choose *shared heap*?

wenyongh, 2024-06-19:
@WenLY1 Many people may not know the background and the design details of the shared heap, I submitted issue #3546 to try to illustrate that, suggest you to post tasks in that issue to track and discuss instead.