wasm-micro-runtime/wasm-micro-runtime #3548: [RFC] New option to do software bounds check by runtime API instead of inline it in IR in AOT/JIT mode

no1wudi, 2024-06-19:
Taking AOT mode as an example, in the current implementation, when performing software boundary checks, wamrc directly expands the boundary check code，please refer to https://github.com/bytecodealliance/wasm-micro-runtime/blob/374653401075abc98e3daf3134a215b993f2839e/core/iwasm/compilation/aot_emit_memory.c#L220-L272

This approach can offer extremely high performance, but the drawback is that it is not very flexible, especially when dealing with multiple memory regions. For example, in scenarios like a shared heap or mmap discussed in https://github.com/bytecodealliance/wasm-micro-runtime/issues/3546.

I believe an additional option can be added，for regular Wasm applications, the current implementation should be maintained to ensure high performance, for shared heap or mmap or other special case, call runtime API (some thing similar to https://github.com/bytecodealliance/wasm-micro-runtime/blob/374653401075abc98e3daf3134a215b993f2839e/core/iwasm/common/wasm_memory.c#L596-L599) to do the check.

**The benefits of this approache**:
1. Implementing complex check logic in C is much simpler and more flexible than using LLVM IR.
2. If there are multiple memory regions, implementing logic similar to if-else if in LLVM IR would significantly increase the size of the final generated code, because each memory access instruction would be expanded into a combination of if-else if statements.

**The drawbacks of this approache**:
1. Calling the runtime API is relatively slower, but if compared to an implementation using if-else if in LLVM IR, it should be acceptable.

yamt, 2024-06-19:
it sounds like a good idea. the aot code size is a real problem for some of use cases.

no1wudi, 2024-06-19:
> it sounds like a good idea. the aot code size is a real problem for some of use cases.

Yes, if you have tried the `wamrc` configuration with `--bounds-check=0`, you will find that the code size is significantly reduced. This scenario can be regarded as a single if-else combination. However, if there are multiple memory regions, more if-else if-else if combinations are required to perform the checks, and it's easy to imagine the huge amount of code that would be generated.