wasm-micro-runtime/wasm-micro-runtime #3553: What's the deference between bound check of fast interpreter mode and bound check of classic interpreter mode

WenLY1, 2024-06-20:
The bound check of classic interpreter mode will do extra work if WASM_ENABLE_MEMORY64 is enabled
https://github.com/bytecodealliance/wasm-micro-runtime/blob/9b9e938bee7b41f026fa9aeee988a5bfc63d55ed/core/iwasm/interpreter/wasm_interp_classic.c#L89-L102

In fast interp mode, bound check does not do extra work  if WASM_ENABLE_MEMORY64 is enabled
https://github.com/bytecodealliance/wasm-micro-runtime/blob/9b9e938bee7b41f026fa9aeee988a5bfc63d55ed/core/iwasm/interpreter/wasm_interp_fast.c#L40-L62

Why the bound check of fast interp mdoe is different from that of classic mode?

TianlongLiang, 2024-07-01:
The fast interp mode hasn't supported the memory64 proposal yet, so the boundary check is unmodified and won't check the overflow condition that only occurs in memory64

WenLY1, 2024-07-03:
> The fast interp mode hasn't supported the memory64 proposal yet, so the boundary check is unmodified and won't check the overflow condition that only occurs in memory64
Thanks, got it