wasm-micro-runtime/wasm-micro-runtime #3568: wamr dont support grain-lang now.

vkensou, 2024-06-25:
Hope wamr support the language: [grain](https://grain-lang.org/)

Now iwasm said: "WASM module load failed: unsupported opcode 12"

the grain code:
```
module Hello

print("Hello, world!")
```

cmd:

```
grain compile "hello.gr"
iwasm "hello.gr.wasm"
```

TianlongLiang, 2024-06-25:
Hi, can you upload the grain-generated wasm file? I can take a look at why this happened?

TianlongLiang, 2024-06-25:
Can you try adding this CMake flag(-DWAMR_BUILD_TAIL_CALL=1) when building WAMR to see if it helps?

vkensou, 2024-06-25:
> Can you try adding this CMake flag(-DWAMR_BUILD_TAIL_CALL=1) when building WAMR to see if it helps?

Add TAIL_CALL is ok! 

[hello.gr.zip](https://github.com/user-attachments/files/15965772/hello.gr.zip)

closed 2024-06-25 by vkensou