wasm-micro-runtime/wasm-micro-runtime #3580: buffer overflow in wasm_loader_emit_br_info

yamt, 2024-07-01:
a crash file for wasm-mutator-fuzz:
[crash-27220c0abb57efbe2e501a90e139b1331a5f71d6.gz](https://github.com/user-attachments/files/16047301/crash-27220c0abb57efbe2e501a90e139b1331a5f71d6.gz)

```
INFO: Running with entropic power schedule (0xFF, 100).
INFO: Seed: 1007482135
INFO: Loaded 1 modules   (31916 inline 8-bit counters): 31916 [0x10795e39f, 0x10796604b), 
INFO: Loaded 1 PC tables (31916 PCs): 31916 [0x107966050,0x1079e2b10), 
./wasm_mutator_fuzz: Running 1 inputs 1 time(s) each.
Running: crash-27220c0abb57efbe2e501a90e139b1331a5f71d6
=================================================================
==45439==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x60600000049c at pc 0x00010776f750 bp 0x7ff7b89e7690 sp 0x7ff7b89e7688
READ of size 2 at 0x60600000049c thread T0
    #0 0x10776f74f in wasm_loader_emit_br_info wasm_loader.c:9267
    #1 0x10776cc17 in wasm_loader_check_br wasm_loader.c:10311
    #2 0x10775d92a in check_branch_block wasm_loader.c:10439
    #3 0x107712464 in wasm_loader_prepare_bytecode wasm_loader.c:11674
    #4 0x1076d8a23 in load_from_sections wasm_loader.c:6126
    #5 0x1076e0556 in load wasm_loader.c:6532
    #6 0x1076df8f1 in wasm_loader_load wasm_loader.c:6709
    #7 0x107772b32 in wasm_load wasm_runtime.c:65
    #8 0x1075ed252 in wasm_runtime_load_ex wasm_runtime_common.c:1356
    #9 0x1075edab7 in wasm_runtime_load wasm_runtime_common.c:1401
    #10 0x107513b0f in LLVMFuzzerTestOneInput wasm_mutator_fuzz.cc:34
    #11 0x10784c4e2 in fuzzer::Fuzzer::ExecuteCallback(unsigned char const*, unsigned long) FuzzerLoop.cpp:612
    #12 0x107835d43 in fuzzer::RunOneTest(fuzzer::Fuzzer*, char const*, unsigned long) FuzzerDriver.cpp:324
    #13 0x10783b75b in fuzzer::FuzzerDriver(int*, char***, int (*)(unsigned char const*, unsigned long)) FuzzerDriver.cpp:860
    #14 0x10786b812 in main FuzzerMain.cpp:20
    #15 0x10b99c52d in start+0x1cd (dyld:x86_64+0x552d) (BuildId: 10c8ed2759df36b5ab457a381b38478332000000200000000100000000070c00)

0x60600000049c is located 4 bytes to the left of 64-byte region [0x6060000004a0,0x6060000004e0)
allocated by thread T0 here:
    #0 0x107fed080 in wrap_malloc+0xa0 (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x47080) (BuildId: ad6ad699f2963cdb985f84296d61f6e42400000010000000000a0a0000010d00)
    #1 0x10751d074 in os_malloc posix_malloc.c:11
    #2 0x1075c1a50 in wasm_runtime_malloc_internal wasm_memory.c:209
    #3 0x1075c18e0 in wasm_runtime_malloc wasm_memory.c:294
    #4 0x1076ec476 in loader_malloc wasm_loader.c:363
    #5 0x10774a137 in wasm_loader_ctx_init wasm_loader.c:8211
    #6 0x107706906 in wasm_loader_prepare_bytecode wasm_loader.c:10957
    #7 0x1076d8a23 in load_from_sections wasm_loader.c:6126
    #8 0x1076e0556 in load wasm_loader.c:6532
    #9 0x1076df8f1 in wasm_loader_load wasm_loader.c:6709
    #10 0x107772b32 in wasm_load wasm_runtime.c:65
    #11 0x1075ed252 in wasm_runtime_load_ex wasm_runtime_common.c:1356
    #12 0x1075edab7 in wasm_runtime_load wasm_runtime_common.c:1401
    #13 0x107513b0f in LLVMFuzzerTestOneInput wasm_mutator_fuzz.cc:34
    #14 0x10784c4e2 in fuzzer::Fuzzer::ExecuteCallback(unsigned char const*, unsigned long) FuzzerLoop.cpp:612
    #15 0x107835d43 in fuzzer::RunOneTest(fuzzer::Fuzzer*, char const*, unsigned long) FuzzerDriver.cpp:324
    #16 0x10783b75b in fuzzer::FuzzerDriver(int*, char***, int (*)(unsigned char const*, unsigned long)) FuzzerDriver.cpp:860
    #17 0x10786b812 in main FuzzerMain.cpp:20
    #18 0x10b99c52d in start+0x1cd (dyld:x86_64+0x552d) (BuildId: 10c8ed2759df36b5ab457a381b38478332000000200000000100000000070c00)

SUMMARY: AddressSanitizer: heap-buffer-overflow wasm_loader.c:9267 in wasm_loader_emit_br_info
Shadow bytes around the buggy address:
  0x1c0c00000040: 00 00 00 00 00 00 06 fa fa fa fa fa 00 00 00 00
  0x1c0c00000050: 00 00 00 00 fa fa fa fa 00 00 00 00 00 00 00 fa
  0x1c0c00000060: fa fa fa fa 00 00 00 00 00 00 00 00 fa fa fa fa
  0x1c0c00000070: 00 00 00 00 00 00 07 fa fa fa fa fa 00 00 00 00
  0x1c0c00000080: 00 00 07 fa fa fa fa fa 00 00 00 00 00 00 07 fa
=>0x1c0c00000090: fa fa fa[fa]00 00 00 00 00 00 00 00 fa fa fa fa
  0x1c0c000000a0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x1c0c000000b0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x1c0c000000c0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x1c0c000000d0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x1c0c000000e0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==45439==ABORTING
```

lum1n0us, 2024-07-02:
May need to re-upload the corpus.

[27220c0abb57efbe2e501a90e139b1331a5f71d6.gz](https://github.com/user-attachments/files/16047301/crash-27220c0abb57efbe2e501a90e139b1331a5f71d6.gz) seems an empty one. `tar xf` and `tar tf` show nothing.

And there is no such a file at https://github.com/yamt/toywasm-fuzzer-corpus/blob/master/corpus/27220c0abb57efbe2e501a90e139b1331a5f71d6

wenyongh, 2024-07-02:
@lum1n0us I can use gunzip to unzip the .gz file and reproduce the issue. The issue is caused by I will submit another PR to fix it.

wenyongh, 2024-07-02:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/3588

yamt, 2024-07-02:
i think our fuzz target should not enable both of fast interpreter and simd because the combination is not well supported. how do you think?

lum1n0us, 2024-07-03:
YES. Currently, WAMR doesn't support fast-interpreter + SIMD.

yamt, 2024-07-10:
fixed by https://github.com/bytecodealliance/wasm-micro-runtime/pull/3588

closed 2024-07-10 by yamt