wasm-micro-runtime/wasm-micro-runtime #3581: Announcement: Transition from TSC Meetings to Monthly Community Meetings

lum1n0us, 2024-07-01:
Hello everyone!

We have some exciting news to share with you all. In an effort to better align with the Bytecode Alliance's requirements for core projects and to foster a more inclusive environment, we're transforming our regular TSC (Technical Steering Committee) meetings into broader community meetings.

Here's what you need to know:

- Frequency: These community meetings will take place once a month, ensuring that everyone has the opportunity to participate and stay updated with the latest developments.

- Duration: Each meeting will last one hour, packed with discussions and updates to make the most of our time together.

- Open to Adjustments: We're open to the idea of holding these meetings twice a month if we find that there's a high level of interest and it becomes challenging to accommodate everyone in a single session.

- Preparation is Key: To ensure our meetings are efficient and productive, we'll be collecting topics in advance. This will help us stick to the agenda and cover all the important points.

- Stay Informed: The meeting minutes and the agenda will be available in a [GitHub repo](https://github.com/bytecodealliance/meetings/tree/main/wamr), so you can always stay in the loop and contribute to the discussions, whether you're able to attend or not.

We encourage a wider audience to join these meetings. Whether you're a long-time contributor or new to the community, your voice and input are valuable to us. Let's collaborate to make our project more robust and welcoming!

Stay tuned for more details on the date and time of our first community meeting. We can't wait to see you there!

lum1n0us, 2024-07-01:
I am going to start a pre-meeting to discuss and set-up few regular things and collect ideas about short-term(half-year)/long-term(1 year) roadmap.

Here is the link to pick up an appropriate time slot. I reused previous TSC meeting schedule. But I am open to other suggestions(about time).

https://doodle.com/meeting/participate/id/dN5qByDb

loganek, 2024-07-01:
Hi @lum1n0us, thanks a lot for setting this up. I think it's a really good idea to provide a forum for everybody to discuss. I have a question about the TSC; will there be still a dedicated TSC meeting given the broader responsibility of TSC itself as per https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/TSC_Charter.md? Thanks

lum1n0us, 2024-07-05:
@wenyongh @xwang98 please feel free to jump in and correct me.

I think we'll have TSC meetings as required. Like making the final technical decision. The community meeting is a open place to share and listen. The TSC meeting is, on the opposite, to vote.

lum1n0us, 2024-07-05:
The Pre-meeting will start in 8th. July. Here is the meeting link. https://bytecodealliance.zulipchat.com/#narrow/stream/290350-wamr/topic/Announcement.3A.20Transition.20from.20TSC.20Meetings.20to.20Monthly.20Commu/near/449186296.

wenyongh, 2024-07-05:
> @wenyongh @xwang98 please feel free to jump in and correct me.
> 
> I think we'll have TSC meetings as required. Like making the final technical decision. The community meeting is a open place to share and listen. The TSC meeting is, on the opposite, to vote.

Agree to have both community meetings and TSC meetings, and yes we can listen and collect the requirements in the community meeting, share what the community has recently accomplished and what we are planning to do, and answer questions from developers. For TSC meetings, maybe each TSC member can share their roadmap if he want, and we can collect some topics previously and discuss them, and focus more on the key features/issues, and also vote.