wasm-micro-runtime/wasm-micro-runtime #3598: iwasm fails on try catch

bashor, 2024-07-05:
### Subject of the issue

`iwasm` on simple example with try catch

### Test case

A bit modified version of [kotlin-wasm-examples/wasi-example](https://github.com/Kotlin/kotlin-wasm-examples/tree/main/wasi-example)

### Steps to reproduce

Tell us how to reproduce this bug or defect.

1. Download and unzip [kotlin-wasm-wasi-example-wasm-wasi.wasm.zip](https://github.com/user-attachments/files/16111213/kotlin-wasm-wasi-example-wasm-wasi.wasm.zip)
2. Build iwasm with GC and EH support.
3. Run `./iwasm kotlin-wasm-wasi-example-wasm-wasi.wasm`.

### Expected output
```
Hello from Kotlin via WASI
OK Hi
Current 'realtime' timestamp is: 1720191249448619000
Current 'monotonic' timestamp is: 4473162847494166
```

### Actual output

```
Hello from Kotlin via WASI
WASM module instantiate failed: Exception: unsupported opcode
```

### Extra Info

Anything else you'd like to add?

bashor, 2024-07-05:
JFTR: it seems to be working when the fast interpreter is disabled

yamt, 2024-07-07:
> JFTR: it seems to be working when the fast interpreter is disabled

EH is implemented only for classic interpreter.