wasm-micro-runtime/wasm-micro-runtime #3599: iwasm doesn't support new/final instructions for Exception Handling proposal

bashor, 2024-07-05:
### Test case

A bit modified version of [kotlin-wasm-examples/wasi-example](https://github.com/Kotlin/kotlin-wasm-examples/tree/main/wasi-example)

### Steps to reproduce

1. Download and unzip [kotlin-wasm-wasi-example-wasm-wasi.wasm.zip](https://github.com/user-attachments/files/16111311/kotlin-wasm-wasi-example-wasm-wasi.wasm.zip)
2. Build iwasm with GC and EH support.
3. Run `./iwasm kotlin-wasm-wasi-example-wasm-wasi.wasm`.

### Expected output
```
Hello from Kotlin via WASI
OK Hi
Current 'realtime' timestamp is: 1720191249448619000
Current 'monotonic' timestamp is: 4473162847494166
```

### Actual output

```
WASM module load failed: unsupported opcode 1f
```

wenyongh, 2024-07-09:
The exception handling proposal has changed a lot after it was implemented in WAMR, it really takes a lot of effort to refactor the code to align with the latest spec. Since the feature was mainly implemented by Siemens guys (see https://github.com/bytecodealliance/wasm-micro-runtime/commit/af318bac818ac36dc7e0641fd6b15b4b52c496d6), I will check with them to see whether they can support it, and discuss how to do next if they have no bandwidth.

bashor, 2024-07-09:
Just in case there is no rush from our (Kotlin) side, we are going to use the old version by default for some time, at least until major browsers support the new version and the majority of users migrate to those browser versions.