wasm-micro-runtime/wasm-micro-runtime #3608: CI doesn't seem to cover xtensa at all

yamt, 2024-07-09:
while we have a considerable amount of xtensa-specific code, the CI doesn't seem to cover it at all.
it would be nice to have basic tests at least.

an obstacle is that we use a different version of llvm for xtensa.

HongxiaWangSSSS, 2024-07-15:
Hi, @yamt -san

I'd like to know do you have any plan to support xtensa platform based on the current CI.
Currently QEMU supports [esp32s3](https://github.com/espressif/qemu/releases/tag/esp-develop-9.0.0-20240606), and may can help support aot and wasm testing (I haven't tried it yet), But it seems that XIP aot cannot be run either.
We are also very interested in the standard test of this platform, especially XIP~

BR,
Hongxia

yamt, 2024-07-16:
> Hi, @yamt -san
>
> I'd like to know do you have any plan to support xtensa platform based on the current CI. Currently QEMU supports [...]

i was able to run (at least some parts of) the spec test with test_wamr.sh with esp32s3 qemu locally while ago.
with manually compiled wamrc and nuttx image.
i guess i will submit a few patches sooner or later.

i haven't done anything wrt the "build llvm+wamrc for xtena on the CI" part.

closed 2024-07-26 by wenyongh