wasm-micro-runtime/wasm-micro-runtime #3614: This value might overflow 64 bit int

g0djan, 2024-07-10:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/1b1ec715e9b16c23f146333de846fb0285ec2401/core/iwasm/libraries/lib-socket/src/wasi/wasi_socket_ext.c#L582

loganek, 2024-07-10:
Thanks, I think this value indeed can overflow, but I'm not sure there's much we can do about it - we could potentially saturate the value, but given how the function is used, and the fact that uint64_t can fit 500,000 years, I don't think this is a real concern.

yamt, 2024-07-17:
> Thanks, I think this value indeed can overflow, but I'm not sure there's much we can do about it - we could potentially saturate the value, but given how the function is used, and the fact that uint64_t can fit 500,000 years, I don't think this is a real concern.

it's a real concern.
consider that a user specified a very long timeout and it actually times out almost immediately because of the overflow.