wasm-micro-runtime/wasm-micro-runtime #3615: Compiled AOT causes SIG 7 Error

bpnetguy, 2024-07-10:
### Compiled AOT causes SIG 7 Error

If I create a global static struct with vector, and insert data  to that vector, I get a sig 7 error. 
If I move static struct in side the init function, all else works fine. 
All this works fine in JIT compilation, but not AOT

### Your environment

* Host OS: ArmV7
* WAMR version: WAMR-2.1.0
* platform: armv7
 * running mode: AOT

### Steps to reproduce
Compile the below as AOT and run it. 

Broken code
```
#include <vector>
struct  T {
        std::vector<int> updates;
};
T t;
__attribute__((export_name("Init"))) int Init() {
    int size = t.updates.size();
    t.updates.push_back(1);
    size = t.updates.size();
    return 0;
}```

This works
```
#include <vector>
struct  T {
        std::vector<int> updates;
};
__attribute__((export_name("Init"))) int Init() {
   T t;
    int size = t.updates.size();
    t.updates.push_back(1);
    size = t.updates.size();
    return 0;
}
```

### Expected behavior

No sig 7 crash

### Actual behavior

Sig 7 crash
### Extra Info

wenyongh, 2024-07-11:
@bpnetguy I used wasi-sdk-22.0 to compile the source code and then compiled the wasm into aot file with wamrc, but no error was found, no matter `T t` is put inside the function or as a global variable. My platform is Ubuntu-22.0. Not sure which platform you use wamrc? And could you upload the wasm file? Thanks.

wenyongh, 2024-07-11:
BTW, I used `/opt/wasi-sdk/bin/clang++ -O3 -o test.wasm test.cpp -pthread --target=wasm32-wasi-threads -Wl,--allow-undefined` to compile the cpp file.

bpnetguy, 2024-07-17:
I figured out the issue. It was related to ARMv7. I had to compile it with `--target=armv7 --cpu=generic --cpu-features=-neon` other wise std vector breaks on armv7

closed 2024-07-17 by bpnetguy