wasm-micro-runtime/wasm-micro-runtime #3621: Slow AOT compilation

bnason-nf, 2024-07-12:
On my machine it takes about 18 minutes for wamrc to compile this wasm file:

https://github.com/bnason-nf/karting/raw/main/wgpukarting.wasm.gz

My command line is:

`wamrc --target=x86_64 -o wgpukarting.aot wgpukarting.wasm`

Even though this wasm file is very large, 18 minutes seems like an unexpectedly long time to compile. I'm building on a fast Linux machine:

```
> lscpu
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         52 bits physical, 57 bits virtual
  Byte Order:            Little Endian
CPU(s):                  48
  On-line CPU(s) list:   0-47
Vendor ID:               AuthenticAMD
  Model name:            AMD Ryzen Threadripper 7960X 24-Cores
    CPU family:          25
    Model:               24
    Thread(s) per core:  2
    Core(s) per socket:  24
    Socket(s):           1
    Stepping:            1
    CPU max MHz:         7786.0000
    CPU min MHz:         545.0000
    BogoMIPS:            8386.83
```

Is there anything that could be done to speed up the compilation?

cngzhnp, 2024-07-12:
Maybe, you can shrink your WASM file by using wasm-opt tool like below:

wasm-opt --dce -Oz -o wgpukarting_output.wasm wgpukarting.wasm 

In my environment, WASM file size reduced form 53.3 MB to 47.3 MB which makes it a little bit faster this. Do you have any compile options how you create the WASM file? Maybe it might be shrinked more than that.

bnason-nf, 2024-07-12:
Thanks @cngzhnp, unfortunately that didn't affect the wamrc compilation time at all.

bnason-nf, 2024-07-12:
Maybe my expectations are wrong, but I would have thought that compilation could happen at least an order of magnitude faster.

yamt, 2024-07-16:
> On my machine it takes about 18 minutes for wamrc to compile this wasm file:
> 
> https://github.com/bnason-nf/karting/raw/main/wgpukarting.wasm.gz

just curious; is this a real application which is expected to run on WAMR?

as this module seems to contain a lot of functions, i guess it's relatively easy to speed up compilation
using more threads as we do for JIT. (well, if your system have spare CPUs, of course.)

bnason-nf, 2024-07-16:
Yes, this is an example one type of application that we would like to be able to run.