wasm-micro-runtime/wasm-micro-runtime #3622: Use a newer version of LLDB for debugging?

iKlask, 2024-07-12:
So it seems that debugging wasm code in WAMR relies on an older version of LLDB (v13.x) with a [specific patch applied](https://reviews.llvm.org/D78801).

Having to pull an older version of LLVM, apply this patch, and build it yourself is quite a hassle, especially since WAMR doesn't supply prebuilt binaries for Windows (only for Linux and macOS). Moreover, it seems unlikely that this patch will ever be integrated into the main LLVM branch.

So my question: Have modern versions of LLVM/LLDB supplied features this patch attempted to add, or is this patch is still necessary today for debugging WASM in WAMR? I am asking because I've noticed that the Wasmtime runtime works well with more modern builds of LLDB without requiring any patches.

yamt, 2024-08-27:
> So my question: Have modern versions of LLVM/LLDB supplied features this patch attempted to add, or is this patch still necessary today for debugging WASM in WAMR?

the latter.

>  I am asking because I've noticed that the Wasmtime runtime works well with more modern builds of LLDB without requiring any patches.

wamr aot-debug doesn't require the patch. (it has its own problems though)

srberard, 2025-08-13:
I'd like to revisit this topic.  As pointed out above, we still need the patch and it's based on a version of LLVM that is several years old.  It would be ideal to not have to have a patch to support debugging interpreted Wasm code.  Also mentioned above, this works out of the box with WasmTime today with no need for a patched version of LLDB.  I'm not sure how this is done in WasmTime, but it proves that this can be done and the experience is excellent.

My desire here is to improve the developer experience and have a good debugging experience.  While I am able to get the existing debugger built and working, I've hit a number of issues and bugs.  Fixing them on the old codebase doesn't seem to be the right thing to do.  

From what I understand, a key issue was with the fact that Wasm supports multiple memories and that this was not handled well in LLDB at the time.  Looking through the patch in the WAMR codebase, if I'm understanding this correctly, a plugin was created for handling this but that required a number of changes in core parts of LLDB.  In the end, it was never upstreamed into LLVM.  Now this was all done a while ago (LLVM 13.x if memory serves) and a lot has changed since then.  I've seen several active issues around Wasm debugging support in the LLVM project including [this](https://github.com/llvm/llvm-project/issues/150449) issue and [this](https://github.com/llvm/llvm-project/issues/151439) issue.  There's clearly interest in making this work.

Would it be possible to revisit this and see what can be done to align things so that debugging works out of the box with WAMR and LLDB?

JDevlieghere, 2026-02-17:
ToT LLDB supports debugging Wasm running under `iwasm` without any patches. I gave a talk about this at the [LLVM Devroom at FOSDEM](https://fosdem.org/2026/schedule/event/87ZLQV-wasm-debugging-lldb/). We should be able to remove the old instructions once LLVM 22 is released.