wasm-micro-runtime/wasm-micro-runtime #3624: Use WAMR as handler in crun runtime

macko99, 2024-07-13:
Hello, I try to add support for wasm-micro-runtime as another handler in [crun](https://github.com/containers/crun). It seems promising since it is a very lightweight WASM runtime.

I have followed the implementation of current crun handlers as well as docs regarding embedding WAMR, but unfortunately, I failed to succeed. Here is the [PR](https://github.com/containers/crun/pull/1497/files) with my efforts.

Would someone be able to help me fix it? Currently, it seems that the wasm module is not executed, regarding the "Running" status reported by kubectl.

I'm using Kubernetes 1.27, containerd 1.6, latest build of crun and WAMR.

closed 2024-07-16 by macko99

wenyongh, 2024-07-16:
@macko99 thanks for advocating WAMR in crun! It now basically works, right?

I tried to look into the PR and have several suggestions: (1) had better check the return value for most of the API calls, (2) should free the buffer read by `read_wasm_binary_to_buffer` after module is unloaded? (3) had better check whether an exception was thrown after `wasm_runtime_call_wasm`, and handle the "wasi proc exit" exception, refer to: https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_application.c#L124-L151 (4) had better set `heap_size` to 0 but not 8096, and export malloc/free functions when compiling wasm app (add `-Wl,--export=malloc -Wl,--export=free` for clang) to avoid using host managed heap and use libc heap directly. Thanks.

macko99, 2025-01-27:
Thank you all for pushing crun+wamr forward! Sadly I had no time in September to finalize the PR, since I was finalizing my master's thesis on the topic. If anyone is interested in memory footprint evaluation: https://atlarge-research.com/pdfs/2024-mkozub-msc_thesis.pdf