wasm-micro-runtime/wasm-micro-runtime #3627: Rust sdk does not work, cannot create an instance "thread signal env initialized failed"

macko99, 2024-07-13:
Hello, would you be able to help me with creating WAMR containerd shim?

I have created some [PR](https://github.com/containerd/runwasi/pull/642) but I end up with one of the errors:

-"thread signal env initialized failed"
without any error
-"error running start function: Failed to create instance: InstantiationFailure("error while pre-opening mapped directory: invalid map"
-"Running "_start"" then "Error: ExecutionError("Exception: unreachable")"
-"Running "_start"" then "error running start function: Failed to find function: FunctionNotFound"

My module has only the main function defined.
I have tried disabling hardware checks in WAMR since I'm working on the VM, [commit](https://github.com/macko99/wamr-rust-sdk/commit/b89f685eba764fd784be103a7659fe19bb215482)

lum1n0us, 2024-07-14:
> error while pre-opening mapped directory: invalid map

means the wasi argument isn't set correctly. if using iwasm, you might want to use `--dir` or `--map-dir` to config WASI.

macko99, 2024-07-14:
@lum1n0us I'm using the [wamr-rust-sdk ](https://github.com/bytecodealliance/wamr-rust-sdk)

lum1n0us, 2024-07-14:
Please refer to 

https://github.com/bytecodealliance/wamr-rust-sdk/blob/main/examples/wasi-hello/src/main.rs#L19-#L21

macko99, 2024-07-14:
@lum1n0us please take a look at the [PR](https://github.com/containerd/runwasi/pull/642/files#diff-91800d108be73643bcffa3d3b8365792b84990e9ed1ad778aa32ccdadf1bad0e)

I do set 
`let wasi_ctx = WasiCtxBuilder::new()
        .set_pre_open_path(vec!["/"], vec!["/"])
        .set_env_vars(envs.iter().map(String::as_str).collect())
        .build();`

I have also tried with other dirs, just like in an example, it fails with "Error: ExecutionError("Exception: unreachable")"

To clarify:
-when setting envs by `set_env_vars` the shim exists with no error or "thread signal env initialized failed"
-when setting `set_pre_open_path` to "/:/" I get "error running start function: Failed to create instance: InstantiationFailure("error while pre-opening mapped directory: invalid map"
-when setting `set_pre_open_path` to ".:" I get "Error: ExecutionError("Exception: unreachable")"