wasm-micro-runtime/wasm-micro-runtime #3674: Run WASM binary in the rodata section

donghengqaz, 2024-07-30:
#### Feature

We are using the WAMR v1.3.x version, and see that the WASM binary can only be put in the data section, or to be more precise the WASM binary is only able to be put into RAM instead of ROM.
﻿
In some low-cost hardware solutions, the RAM is too low to run multiple WASM binary, and I also try to XIP-AOT, but its size looks like much larger than the WASM binary(some AOT data also needs to be put into RAM?). Because they are low-cost hardware solutions, so ROM space should also be saved. 
﻿
Is there any method to run the WASM binary in the rodata section? Or do you have any plans for this?
﻿
#### Benefit

When this feature is implemented, the low-cost hardware solution can save much RAM for the applications and run more WASM binary at the same time.

#### Implementation

No implementation, but if I could do something, please tell me.

#### Alternatives

No.

yamt, 2024-08-01:
>  Is there any method to run the WASM binary in the rodata section? Or do you have any plans for this?

i don't think wamr provides a suitable running mode for the use case right now.
i'm not aware of anyone working on it either.

maybe it isn't too difficult to make the classic interpreter avoid in-place modifications.
it might be a bit cumbersome to deal with value-polymorphic instructions though.
cf. https://github.com/yamt/toywasm/blob/master/doc/annotations.md#type-annotations-for-value-polymorphic-instructions

sasq64, 2024-11-14:
It is actually possible to run the FAST interpreter from ROM. It uses more memory though.