wasm-micro-runtime/wasm-micro-runtime #3680: memory may not be shared: threads not allowed

subhakr, 2024-08-01:
Hi,
here i am trying integrate wasm with sysbench benchmark profiling tool.
i have compiled this sysbench with wasi-sdk 23 version.
after compiling i am trying to run with iwasm runtime.

i am getting like this and my executable name is sysbench.
* Host OS : ubuntu22.04LTS
* WAMR version: iwasm 2.1.1 
* Platform : x86_64

```
admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/sysbench_main$ /home/admin1/Downloads/runtimes_wasm/iwasm2 src/sysbench
WASM module load failed: invalid section id
admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/sysbench_main$ wasm-validate src/sysbench
0000a7f: error: memory may not be shared: threads not allowed

```
```
am__is_gnu_make = { \
  if test -z '$(MAKELEVEL)'; then \
    false; \
  elif test -n '$(MAKE_HOST)'; then \
    true; \
  elif test -n '$(MAKE_VERSION)' && test -n '$(CURDIR)'; then \
    true; \
  else \
    false; \
  fi; \
}
am__make_running_with_option = \
  case $${target_option-} in \
      ?) ;; \
      *) echo "am__make_running_with_option: internal error: invalid" \
              "target option '$${target_option-}' specified" >&2; \
         exit 1;; \
  esac; \
  has_opt=no; \
  sane_makeflags=$$MAKEFLAGS; \
  if $(am__is_gnu_make); then \
    sane_makeflags=$$MFLAGS; \
  else \
    case $$MAKEFLAGS in \
      *\\[\ \	]*) \
        bs=\\; \
        sane_makeflags=`printf '%s\n' "$$MAKEFLAGS" \
          | sed "s/$$bs$$bs[$$bs $$bs	]*//g"`;; \
    esac; \
  fi; \
  skip_next=no; \
  strip_trailopt () \
  { \
    flg=`printf '%s\n' "$$flg" | sed "s/$$1.*$$//"`; \
  }; \
  for flg in $$sane_makeflags; do \
    test $$skip_next = yes && { skip_next=no; continue; }; \
    case $$flg in \
      *=*|--*) continue;; \
        -*I) strip_trailopt 'I'; skip_next=yes;; \
      -*I?*) strip_trailopt 'I';; \
        -*O) strip_trailopt 'O'; skip_next=yes;; \
      -*O?*) strip_trailopt 'O';; \
        -*l) strip_trailopt 'l'; skip_next=yes;; \
      -*l?*) strip_trailopt 'l';; \
      -[dEDm]) skip_next=yes;; \
      -[JT]) skip_next=yes;; \
    esac; \
    case $$flg in \
      *$$target_option*) has_opt=yes; break;; \
    esac; \
  done; \
[... 782 lines cut ...]
uninstall-am: uninstall-binPROGRAMS

.MAKE: $(am__recursive_targets) install-am install-strip

.PHONY: $(am__recursive_targets) CTAGS GTAGS TAGS all all-am \
	am--depfiles check check-am clean clean-binPROGRAMS \
	clean-generic clean-libtool cscopelist-am ctags ctags-am \
	distclean distclean-compile distclean-generic \
	distclean-libtool distclean-tags distdir dvi dvi-am html \
	html-am info info-am install install-am install-binPROGRAMS \
	install-data install-data-am install-dvi install-dvi-am \
	install-exec install-exec-am install-html install-html-am \
	install-info install-info-am install-man install-pdf \
	install-pdf-am install-ps install-ps-am install-strip \
	installcheck installcheck-am installdirs installdirs-am \
	maintainer-clean maintainer-clean-generic mostlyclean \
	mostlyclean-compile mostlyclean-generic mostlyclean-libtool \
	pdf pdf-am ps ps-am tags tags-am uninstall uninstall-am \
	uninstall-binPROGRAMS

.PRECIOUS: Makefile
```
Please check the makefile and suggest me to solve this issue.
or do i need to add any extra flags for my iwasm.
Thanks in advance.

yamt, 2024-08-01:
i guess you need to enable at least:
-DWAMR_BUILD_LIB_WASI_THREADS=1
-DWAMR_BUILD_LIBC_WASI=1

subhakr, 2024-08-01:
> i guess you need to enable at least: -DWAMR_BUILD_LIB_WASI_THREADS=1 -DWAMR_BUILD_LIBC_WASI=1

```
make CFLAGS="-O3 -funroll-loops --sysroot=/home/admin1/Downloads/wasi-sdk-23.0-x86_64-linux/share/wasi-sysroot -pthread -fexceptions -D_WASI_EMULATED_PROCESS_CLOCKS -DWAMR_BUILD_LIB_WASI_THREADS=1 -DWAMR_BUILD_LIBC_WASI=1 -matomics -mbulk-memory" LDFLAGS="--sysroot=/home/admin1/Downloads/wasi-sdk-23.0-x86_64-linux/share/wasi-sysroot \
              -pthread -fexceptions -Wl,--max-memory=1769472 -z stack-size=32768 \
              -Wl,--shared-memory -g -lwasi-emulated-mman -Wl,--export-all \
              -Wl,--no-entry,--strip-all -Wl,--export=__main_argc_argv \
              -Wl,--export=__heap_base -Wl,--export=__data_end \
              -pthread -lwasi-emulated-process-clocks"

```
I have included like this in my make command
but still i am getting this issue
```
admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/sysbench_main$ /home/admin1/Downloads/runtimes_wasm/iwasm2 src/sysbench
WASM module load failed: invalid section id
admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/sysbench_main$ wasm-validate src/sysbench
0000db5: error: memory may not be shared: threads not allowed
```

suggest me if i am doing wrong

and this is my linking stage
```
libtool: link: /home/admin1/Downloads/wasi-sdk-23.0-x86_64-linux/bin/clang-18 --target=wasm32-wasi-threads -Wall -Wextra -Wpointer-arith -Wbad-function-cast -Wstrict-prototypes -Wnested-externs -Wno-format-zero-length -Wundef -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wredundant-decls -Wcast-align -Wvla -O3 -funroll-loops --sysroot=/home/admin1/Downloads/wasi-sdk-23.0-x86_64-linux/share/wasi-sysroot -pthread -fexceptions -D_WASI_EMULATED_PROCESS_CLOCKS -DWAMR_BUILD_LIB_WASI_THREADS=1 -DWAMR_BUILD_LIBC_WASI=1 -matomics -mbulk-memory --sysroot=/home/admin1/Downloads/wasi-sdk-23.0-x86_64-linux/share/wasi-sysroot -pthread -fexceptions -Wl,--max-memory=1769472 -z stack-size=32768 -Wl,--shared-memory -g -Wl,--export-all -Wl,--no-entry -Wl,--strip-all -Wl,--export=__main_argc_argv -Wl,--export=__heap_base -Wl,--export=__data_end -pthread -o sysbench sysbench.o db_driver.o sb_timer.o sb_options.o sb_logger.o sb_histogram.o sb_rand.o sb_thread.o sb_barrier.o sb_lua.o sb_util.o sb_counter.o  -lwasi-emulated-mman -lwasi-emulated-process-clocks tests/fileio/libsbfileio.a tests/threads/libsbthreads.a tests/memory/libsbmemory.a tests/cpu/libsbcpu.a tests/mutex/libsbmutex.a /home/admin1/Downloads/wasi-sdk-23.0-x86_64-linux/share/wasi-sysroot/lib/liblua.a /home/admin1/Downloads/wasi-sdk-23.0-x86_64-linux/share/wasi-sysroot/lib/libck.a -lm -pthread
make[2]: Leaving directory '/home/admin1/sysbench_main/src'
make[1]: Leaving directory '/home/admin1/sysbench_main/src'
Making all in tests
make[1]: Entering directory '/home/admin1/sysbench_main/tests'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/admin1/sysbench_main/tests'
make[1]: Entering directory '/home/admin1/sysbench_main'
make[1]: Nothing to be done for 'all-am'.
make[1]: Leaving directory '/home/admin1/sysbench_main'
```

yamt, 2024-08-01:
"-DWAMR_BUILD_LIB_WASI_THREADS=1 -DWAMR_BUILD_LIBC_WASI=1" are build options for wamr.
not your app.

subhakr, 2024-08-01:
> "-DWAMR_BUILD_LIB_WASI_THREADS=1 -DWAMR_BUILD_LIBC_WASI=1" are build options for wamr.

[image]
[image]

and i am compiling with wasi-sdk23  version
from initial stage i have enabled those two flags.

any other suggestions.

yamt, 2024-08-01:
can you share your wasm module?

subhakr, 2024-08-01:
> can you share your wasm module?
[iwasm.zip](https://github.com/user-attachments/files/16456178/iwasm.zip)

subhakr, 2024-08-01:
> i meant src/sysbench.

yeah sure..

[sysbench.zip](https://github.com/user-attachments/files/16456412/sysbench.zip)

yamt, 2024-08-01:
> > i meant src/sysbench.
> 
> yeah sure..
> 
> [sysbench.zip](https://github.com/user-attachments/files/16456412/sysbench.zip)

as this module seems using exception-handling, you need iwasm built with:
-DWAMR_BUILD_FAST_INTERP=0
-DWAMR_BUILD_EXCE_HANDLING=1

also, i guess this module was built with an older setjmp/longjmp translation.
libsetjmp.a shipped with wasi-sdk only supports the latest (unreleased) version of llvm.

subhakr, 2024-08-01:
> as this module seems using exception-handling, you need iwasm built with: -DWAMR_BUILD_FAST_INTERP=0 -DWAMR_BUILD_EXCE_HANDLING=1
>
> also, i guess this module was built with an older setjmp/longjmp [...]

[image]
[image]

After including these flags have the same issue.

and while building this module i did't face any issue with libsetjmp.a.
so my assumption is i don't have issue with this library.

any other suggestions for this issue.

yamt, 2024-08-01:
> > > > i meant src/sysbench.
> > > 
> > > 
> > > yeah sure..
> > > [sysbench.zip](https://github.com/user-attachments/files/16456412/sysbench.zip)
> > 
> > 
> > as this module seems using exception-handling, you need iwasm built with: -DWAMR_BUILD_FAST_INTERP=0 -DWAMR_BUILD_EXCE_HANDLING=1
> > also, i guess this module was built with an older setjmp/longjmp translation. libsetjmp.a shipped with wasi-sdk only supports the latest (unreleased) version of llvm.
> 
> ![Screenshot from 2024-08-01 19-16-23](https://private-user-images.githubusercontent.com/126141203/354249878-17022ed7-1791-4243-a07c-56af5a244c25.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI1MjA3OTAsIm5iZiI6MTcyMjUyMDQ5MCwicGF0aCI6Ii8xMjYxNDEyMDMvMzU0MjQ5ODc4LTE3MDIyZWQ3LTE3OTEtNDI0My1hMDdjLTU2YWY1YTI0NGMyNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwMVQxMzU0NTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NDhiOGQwZjYxNGUzZmNiYjkxNDBhYWQ2NmExM2Y3MzMxOWJiMzcxNzQ3YTI2YmExODQ4MzMyMWVhZmVmOTRlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.JJMgOhwpIbBLeNZbsdWdA2WSyx5K45uzyVe9OZb19vQ) ![image](https://private-user-images.githubusercontent.com/126141203/354250118-4c1ed896-bde0-410a-8b2e-a4127a839c1c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI1MjA3OTAsIm5iZiI6MTcyMjUyMDQ5MCwicGF0aCI6Ii8xMjYxNDEyMDMvMzU0MjUwMTE4LTRjMWVkODk2LWJkZTAtNDEwYS04YjJlLWE0MTI3YTgzOWMxYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwMVQxMzU0NTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMjY0ZWRlNTQyNDhiZjM2MDQwYjMxOGIyY2JjNmFlZDU2NWVmNzIzNmYwZjFiZGNlNDJlZTUzODA2MDYxY2JhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.8zlLYYXNUotWFUuINXZvmlRu-R3xvW4kenvRv7WPLj4)
> 
> After including these flags have the same issue.

have you rebuilt it after changing cmake options?

> 
> and while building this module i did't face any issue with libsetjmp.a. so my assumption is i don't have issue with this library.

your sysbench imports functions like env:saveSetjmp, which is a sign of setjmp issues.

> 
> any other suggestions for this issue.

subhakr, 2024-08-01:
[image]

after running below command while doing make i am getting above error(PFA)
cmake .. -DWAMR_BUILD_FAST_INTERP=0 -DWAMR_BUILD_EXCE_HANDLING=1

i have tried with
cmake .. -DWAMR_BUILD_FAST_INTERP=0 -DWAMR_BUILD_EXCE_HANDLING=0
with this command iwasm is generating without any error
i think exception is not yet implemented.
Am i right?
**and for setjump i am not seeing any issue.For your reference i have attached the log.txt file.**
[log.txt](https://github.com/user-attachments/files/16458810/log.txt)

yamt, 2024-08-01:
> ![image](https://private-user-images.githubusercontent.com/126141203/354287989-fad0a096-f19c-4fb9-a560-6164a42ed7df.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

i don't know what wamr version you are using.
in the latest version it works.

> **and for setjump i am not seeing any issue.For your reference i have attached the log.txt file.** [log.txt](https://github.com/user-attachments/files/16458810/log.txt)

i don't understand how this log is related to the sysbench binary you uploaded. i suspect it's a different module.
anyway, if you don't trust me, it isn't my problem.

subhakr, 2024-08-01:
No i trust you thats why i am asking you to suggest me.

i think i am using latest code from github
without enabling exception , i can see this version
```
admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/wasm-micro-runtime/product-mini/platforms/linux/build$ ./iwasm --version
iwasm 2.1.1
```

- i don't understand how this log is related to the sysbench binary you uploaded. i suspect it's a different module.

sysbench.wasm and sysbench both are same only just i have converted to .wasm from sysbench module

if you don't mind
can i get brief solution from your side regarding setjump like
how can i see those errors and how to solve those issues.

Thanks in advance.

yamt, 2024-08-02:
> No i trust you thats why i am asking you to suggest me.
> 
> yeah with latest one its working for exception flag.
> 
> ```
> admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/Documents/wasm-micro-runtime/product-mini/platforms/linux/build$ ./iwasm --version
> iwasm 2.1.1
> admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/Documents/wasm-micro-runtime/product-mini/platforms/linux/build$ ./iwasm /home/admin1/sysbench_main/src/sysbench
> WASM module load failed: invalid limits flags

have you enabled -DWAMR_BUILD_LIB_WASI_THREADS=1?

> admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/Documents/wasm-micro-runtime/product-mini/platforms/linux/build$ wasm-validate /home/admin1/sysbench_main/src/sysbench
> 0000db5: error: memory may not be shared: threads not allowed
> ```

wasm-validate by default doesn't accept threads or exceptions.

> 
> So now you are saying like problem with setjump library.
> 
> **i don't understand how this log is related to the sysbench binary you uploaded. i suspect it's a different module.**
> 
> sysbench.wasm and sysbench both are same only just i have converted to .wasm from sysbench module

log.txt contains a lot of import resolution errors.
otoh, sysbench module you provided doesn't have those imports. (thus errors in log.txt can't happen)

> 
> if you don't mind can i get brief solution from your side regarding setjump like how can i see those errors and how to solve those issues.

investigate the module (eg. by wasm2wat) and see what it imports.
"saveSetjmp" etc are what's used to implement setjmp/longjmp for the old versions of emscripten.
it isn't compatible with wasi-sdk libsetjmp.a.

here's an example script to build micropython, which is an application using setjmp/longjmp, with wasi-sdk libsetjmp.a.
https://github.com/yamt/micropython/blob/unix-wasi/ports/unix/build-wasi.sh
at this point, you need to use an unreleased version of llvm.

> 
> Thanks in advance.

subhakr, 2024-08-02:
> investigate the module (eg. by wasm2wat) and see what it imports.
[import_functions.txt](https://github.com/user-attachments/files/16465872/import_functions.txt)

yeah i have used wasm-objdump command to see import functions.
just for your reference i have attached this file. From this file only i got to know i am importing setjmp and longjmp functions.

[import_functions.txt](https://github.com/user-attachments/files/16465872/import_functions.txt)

> it isn't compatible with wasi-sdk libsetjmp.a.

So upto get the unreleased version of LLVM means clang-19 that too it should become part for wasi-sdk , i can't move further for sysbench module. 
Am i right?
Orelse any other methods are there to solve this issue.
Please help to solve this issue if it's possible.

**And am i fcaing issue with libsetjmp.a only or any other issues are there with my sysbench module. Pls let me know if its there any.**

yamt, 2024-08-02:
> > investigate the module (eg. by wasm2wat) and see what it imports.
> 
> yeah i have used wasm-objdump command to see import functions. just for your reference i have attached this file. From this file only i got to know i am importing setjmp and longjmp functions.
> 
> [import_functions.txt](https://github.com/user-attachments/files/16465872/import_functions.txt)
> 
> > it isn't compatible with wasi-sdk libsetjmp.a.
> 
> So upto get the unreleased version of LLVM means clang-19 that too it should become part for wasi-sdk , i can't move further for sysbench module. Am i right? Orelse any other methods are there to solve this issue. Please help to solve this issue if it's possible.

you can install llvm-19 by yourself and use it with wasi-sdk. (as it's done in the micropython script shown above.)

if it's too cumbersome for you, maybe you need to wait for a future release of wasi-sdk with llvm-19.
(probably in this summer.)

> 
> **And am i facing issue with libsetjmp.a only or any other issues are there with my sysbench module. Pls let me know if its there any.**

your sysbench module seems dying before actually using setjmp/longjmp.
i guess it has more issues. but i don't know what they are.

```shell
spacetanuki% iwasm sysbench
[16:34:23:509 - 1108F2600]: warning: failed to link import function (env, __wasm_longjmp)
[16:34:23:509 - 1108F2600]: warning: failed to link import function (env, saveSetjmp)
[16:34:23:509 - 1108F2600]: warning: failed to link import function (env, getTempRet0)
[16:34:23:509 - 1108F2600]: warning: failed to link import function (env, testSetjmp)
Exception: out of bounds memory access
spacetanuki% wasm-opt --translate-to-exnref --enable-threads --enable-exception-handling --enable-bulk-memory -o sysbench.exnref sysbench
spacetanuki% toywasm --allow-unresolved-functions --wasi sysbench.exnref
Error: PTHREAD_SCOPE_SYSTEM is enabled so please add pthread_attr_setscope(&sb_thread_attr,PTHREAD_SCOPE_SYSTEM) function subha 
spacetanuki% 
```

subhakr, 2024-08-02:
> ```shell
> spacetanuki% iwasm sysbench
> [16:34:23:509 - 1108F2600]: warning: failed to link import function (env, __wasm_longjmp)
> [16:34:23:509 - 1108F2600]: warning: failed to link import function [...]

before giving preference to llvm-19, i would like to solve the remaing issues.
from above CLI i can see pthread error.
Like this how can i see the remaining errors.
Can i get the CLI commnads to list out the issues with my sysbnech module.

yamt, 2024-08-02:
>  Can i get the CLI commnads to list out the issues with my sysbnech module.

build iwasm with
```shell
cmake -DWAMR_BUILD_LIBC_BUILTIN=0 -DWAMR_BUILD_FAST_INTERP=0 -DWAMR_BUILD_EXCE_HANDLING=1 -DWAMR_BUILD_AOT=0 -DWAMR_BUILD_LIBC_WASI=1 -DWAMR_BUILD_LIB_WASI_THREADS=1 ..
```

```shell
spacetanuki% iwasm sysbench
[17:12:01:491 - 10A6BF600]: warning: failed to link import function (env, __wasm_longjmp)
[17:12:01:491 - 10A6BF600]: warning: failed to link import function (env, saveSetjmp)
[17:12:01:491 - 10A6BF600]: warning: failed to link import function (env, getTempRet0)
[17:12:01:491 - 10A6BF600]: warning: failed to link import function (env, testSetjmp)
Error: PTHREAD_SCOPE_SYSTEM is enabled so please add pthread_attr_setscope(&sb_thread_attr,PTHREAD_SCOPE_SYSTEM) function subha 
spacetanuki% 
```

(it seems you get "out of bounds memory access" instead if you enable -DWAMR_BUILD_MEMORY64=1. i dunno why.)

i'm using macOS. but i guess it's same for linux.

subhakr, 2024-08-02:
Hmm , i can see those errors
gimme some time to resolve these errors.

subhakr, 2024-08-05:
> you can install llvm-19 by yourself and use it with wasi-sdk. (as it's done in the micropython script shown above.)

Hi, i would like install LLVM-19
May I get the steps to install this LLVM-19
as i told you i have resolved my application issues.
now i am getting only following issues.
```
admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/sysbench_main$ /home/admin1/Documents/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm  src/sysbench
[02:58:52:668 - 70933A067740]: warning: failed to link import function (env, __wasm_longjmp)
[02:58:52:668 - 70933A067740]: warning: failed to link import function (env, saveSetjmp)
[02:58:52:668 - 70933A067740]: warning: failed to link import function (env, getTempRet0)
[02:58:52:668 - 70933A067740]: warning: failed to link import function (env, testSetjmp)
Attempting to allocate 1064960 bytes of memory...
sysbench 1.1.0-2ca9e3f (using Lua Lua 5.3)

Exception: failed to call unlinked import function (env, saveSetjmp)
```