wasm-micro-runtime/wasm-micro-runtime #3690: Unable to run C++ program in esp32

Sandesh-verma, 2024-08-06:
I have a hello world program for C++. Then I created a .wasm file for the same using the below command with wasi-sdk.
 
```
/opt/wasi-sdk/bin/clang++ -O3  \
        -z stack-size=4096 -Wl,--initial-memory=65536 \
        -o test.wasm main.cpp \
        -Wl,--export=main -Wl,--export=__main_argc_argv \
        -Wl,--export=__data_end -Wl,--export=__heap_base \
        -Wl,--no-entry -Wl,--allow-undefined \
        -fno-exceptions \
        -nostdlib \
```

 
I have used [build script](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/app-samples/hello-world/build.sh) as reference for creating the wasm and header files (wasm and test_wasm.h are created successfully.)
 
On trying to execute the generated test.wasm, using WAMR, I get the below errors..

```
iwasm test.wasm 

[02:07:06:602 - 7FDB24141740]: failed to check signature '(i)' and resolve pointer params for import function (env, abort)
[02:07:06:602 - 7FDB24141740]: warning: failed to link import function (env, _ZNKSt3__28ios_base6getlocEv)
[02:07:06:602 - 7FDB24141740]: warning: failed to link import function (env, _ZNKSt3__26locale9use_facetERNS0_2idE)
[02:07:06:602 - 7FDB24141740]: warning: failed to link import function (env, _ZNSt3__26localeD1Ev)
[02:07:06:602 - 7FDB24141740]: warning: failed to link import function (env, _ZNSt3__213basic_ostreamIcNS_11char_traitsIcEEE3putEc)
[02:07:06:602 - 7FDB24141740]: warning: failed to link import function (env, _ZNSt3__213basic_ostreamIcNS_11char_traitsIcEEE5flushEv)
[02:07:06:602 - 7FDB24141740]: warning: failed to link import function (env, _ZNSt3__213basic_ostreamIcNS_11char_traitsIcEEE6sentryC1ERS3_)
[02:07:06:602 - 7FDB24141740]: warning: failed to link import function (env, _ZNSt3__28ios_base5clearEj)
[02:07:06:602 - 7FDB24141740]: warning: failed to link import function (env, _ZNSt3__213basic_ostreamIcNS_11char_traitsIcEEE6sentryD1Ev)
[02:07:06:602 - 7FDB24141740]: warning: failed to link import function (env, _Znwm)
[02:07:06:602 - 7FDB24141740]: warning: failed to link import function (env, _ZdlPv)
[02:07:06:602 - 7FDB24141740]: warning: failed to link import function (env, abort)
[02:07:06:602 - 7FDB24141740]: warning: allocate 16 bytes memory failed
Exception: allocate memory failed
 ```
 
I am also facing the same error when loading the test_wasm.h file to [main.c](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/platforms/esp-idf/main/main.c) and trying to build for esp32

yamt, 2024-08-06:
please avoid cross-posting.
https://github.com/WebAssembly/wasi-sdk/issues/470
it it's really necessary, reference each other.

Sandesh-verma, 2024-08-06:
@yamt I had tried the command you have provided [here](https://github.com/WebAssembly/wasi-sdk/issues/470#issuecomment-2270530135) earlier..Though I am able to run the wasm file generated(from C++) successfully, I am unable to build the esp32 example code after including the generated header from the wasm file in [main.c](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/platforms/esp-idf/main/main.c).

```
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated /home/user/wamr-code/wasm-micro-runtime/product-mini/platforms/esp-idf/build/bootloader/bootloader.bin
[107/107] cd /home/user/wamr-code/wasm-micro-runtime/pro...duct-mini/platforms/esp-idf/build/bootloader/bootloader.bin
Bootloader binary size 0x6880 bytes. 0x780 bytes (7%) free.
[1008/1010] Linking CXX executable wamr-simple.elf
FAILED: wamr-simple.elf 
: && /home/user/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240305/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=/home/user/wamr-code/wasm-micro-runtime/product-mini/platforms/esp-idf/build/wamr-simple.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld CMakeFiles/wamr-simple.elf.dir/project_elf_src_esp32.c.obj -o wamr-simple.elf -L/home/user/esp-idf/components/soc/esp32/ld   -L/home/user/esp-idf/components/esp_rom/esp32/ld   -L/home/user/wamr-code/wasm-micro-runtime/product-mini/platforms/esp-idf/build/esp-idf/esp_system/ld   -L/home/user/esp-idf/components/esp_phy/lib/esp32   -L/home/user/esp-idf/components/esp_wifi/lib/esp32 esp-idf/xtensa/libxtensa.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/pthread/libpthread.a  esp-idf/cxx/libcxx.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_driver_gptimer/libesp_driver_gptimer.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/esp_driver_uart/libesp_driver_uart.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_driver_pcnt/libesp_driver_pcnt.a  esp-idf/esp_driver_spi/libesp_driver_spi.a  esp-idf/esp_driver_mcpwm/libesp_driver_mcpwm.a  esp-idf/esp_driver_i2s/libesp_driver_i2s.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_driver_sdmmc/libesp_driver_sdmmc.a  esp-idf/esp_driver_sdspi/libesp_driver_sdspi.a  esp-idf/esp_driver_sdio/libesp_driver_sdio.a  esp-idf/esp_driver_dac/libesp_driver_dac.a  esp-idf/esp_driver_rmt/libesp_driver_rmt.a  esp-idf/esp_driver_sdm/libesp_driver_sdm.a  esp-idf/esp_driver_i2c/libesp_driver_i2c.a  esp-idf/esp_driver_ledc/libesp_driver_ledc.a  esp-idf/driver/libdriver.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_vfs_console/libesp_vfs_console.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_driver_cam/libesp_driver_cam.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/wasm-micro-runtime/libwasm-micro-runtime.a  esp-idf/main/libmain.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_driver_cam/libesp_driver_cam.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/espcoredump/libespcoredump.a  -u esp_system_include_coredump_init  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  -u nvs_sec_provider_inc [... cut ...]
hence I am using the below options (reference: [build.sh](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/app-samples/hello-world/build.sh) ) 
```
/opt/wasi-sdk/bin/clang++ -O3 \
        -z stack-size=4096 -Wl,--initial-memory=65536 \
        -o test.wasm main.cpp \
        -Wl,--export=main -Wl,--export=__main_argc_argv \
        -Wl,--export=__data_end -Wl,--export=__heap_base \
        -Wl,--no-entry \
        -Wl,--allow-undefined \
        -fno-exceptions \
        -nostdlib \
```

yamt, 2024-08-06:
> region `dram0_0_seg' overflowed by 61624 bytes

probably your module is just too large?
have you tried strip, wasm-opt, etc?

Sandesh-verma, 2024-08-06:
I have tried using wasm-opt but the size of the produced wasm binary is the same as the original one. 

```
user@user:~/wamr-code/wasm-micro-runtime/product-mini/app-samples/hello-world$ wasm-opt -Oz -o stripped-test.wasm test.wasm 
user@user:~/wamr-code/wasm-micro-runtime/product-mini/app-samples/hello-world$ du -h
220K	stripped-test.wasm
220K	test.wasm
```

And I face the same error of seg overflow.

yamt, 2024-08-07:
> I have tried using wasm-opt but the size of the produced wasm binary is the same as the original one.
>
> ```
> user@user:~/wamr-code/wasm-micro-runtime/product-mini/app-samples/hello-world$ wasm-opt -Oz [...]

although i don't know what your hello world program is doing,
the size of your wasm file seems reasonable for a C++ program.
if i were you, i'd try "-no-rtti -flto=full -Wl,-mllvm,-enable-merge-functions" with wasi-sdk 24.0.
but i suspect there is no easy way to shrink it much.

Sandesh-verma, 2024-08-13:
@yamt i tried with options "-no-rtti -flto=full -Wl,-mllvm,-enable-merge-functions" with wasi-sdk 24.0 ->  the file size increases maybe due to wasi-sdk 24.0 ,  i still get the same error .

```
$/opt/wasi-sdk/bin/clang++ -O3 -o test.wasm main.cpp -fno-rtti -flto=full -Wl,-mllvm,-enable-merge-functions
$ du -h test.wasm 
1.4M	test.wasm
```
```
	

Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated /home/user/wamr-code/wasm-micro-runtime/product-mini/platforms/esp-idf/build/bootloader/bootloader.bin
[107/107] cd /home/user/wamr-code/wasm-micro-runtime/pro...duct-mini/platforms/esp-idf/build/bootloader/bootloader.bin
Bootloader binary size 0x6880 bytes. 0x780 bytes (7%) free.
[1008/1010] Linking CXX executable wamr-simple.elf
FAILED: wamr-simple.elf 
wamr-simple.elf section `.dram0.data' will not fit in region `dram0_0_seg'
DRAM segment data does not fit.
DRAM segment data does not fit.
section .iram0.vectors LMA [40080000,40080402] overlaps section .dram0.data LMA [3ffb0000,401e454f]
region `dram0_0_seg' overflowed by 2136632 bytes
collect2: error: ld returned 1 exit status

```

also interested in knowing if WAMR supports running wasm files generated using emscripten or any other compliers  for esp32 , or is the support limited to wasm files produced using wasi-sdk

yamt, 2024-08-13:
recent wasi-sdk ships libraries with RelWithDebInfo.
you can get a comparable size binary by stripping it.

my usual advice is "don't use C++". :-)

if your esp32 has external flash, maybe you can place your wasm file there.

Sandesh-verma, 2024-08-13:
@yamt thanks for the quick response . I will look into stripping of RelWithDebInfo library
Any suggestions on-
```
also interested in knowing if WAMR supports running wasm files generated using emscripten or any other compliers for esp32 , or is the support limited to wasm files produced using wasi-sdk
```

yamt, 2024-08-13:
> @yamt thanks for the quick response . I will look into stripping of RelWithDebInfo library Any suggestions on-
>
> ```
> also interested in knowing if WAMR supports running wasm files generated using [...]

there is some support for emscripten. but i don't know its state at all.
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md#build-wasm-applications-with-emsdk

zig has wasi target. i dunno c++ works or not though.