wasm-micro-runtime/wasm-micro-runtime #3729: File operations are not supporting

subhakr, 2024-08-18:
### Subject of the issue

File operations are not supporting for sysbench application

### Test case

I would like do bench mark for fileio operations  in that process i am getting facing issue.
I would like to know WAMR will support file opeartions or not?

### Your environment

* Host OS: Ubuntu 22.04 LTS
* WAMR version: 2.1.1
*  cpu architecture: x86_64

### Steps to reproduce
[sysbench_wasm_module.zip](https://github.com/user-attachments/files/16647298/sysbench_wasm_module.zip)

To reprodure the error run the below command.
/home/admin1/Documents/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm --max-threads=64 src/sysbench fileio --file-total-size=2G --file-test-mode=rndrw prepare
 Replace YOUR iwasm path

### Expected behavior

Files have to create after running this command.

### Actual behavior

```
admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/sysbench_main$ /home/admin1/Documents/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm --max-threads=64 src/sysbench fileio --file-total-size=2G --file-test-mode=rndrw prepare
Attempting to allocate 1064960 bytes of memory...
sysbench 1.1.0-2ca9e3f (using Lua Lua 5.3)

128 files, 16384Kb each, 2048Mb total
Creating files for the test...
Extra file open flags: (none)
FATAL: Can't open file errno = 44 (No such file or directory)
```

TianlongLiang, 2024-08-19:
Hi, you can add the command line option `--dir=<dir>` to grant wasi access to the given host directories to the program, in your case, you can perform file operations  under the certain directory you passed

For more details on how you can use iwasm, you can try
```bash
./iwasm --help
```

subhakr, 2024-08-21:
admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/sysbench_main$ /home/admin1/Documents/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm --max-threads=64 --dir=/home/admin1/sysbench_main/fileio_data src/sysbench fileio --file-total-size=2G --file-test-mode=rndrw prepare
Attempting to allocate 1064960 bytes of memory...
sysbench 1.1.0-2ca9e3f (using Lua Lua 5.3)

128 files, 16384Kb each, 2048Mb total
Creating files for the test...
Extra file open flags: (none)
FATAL: Can't open file errno = 44 (No such file or directory)

After doing like that, still i am facing the same issue.
like below
admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/sysbench_main$ ls
aclocal.m4  autom4te.cache  a.worker.js  config      config.status  configure~    COPYING  Dockerfile   install-sh  m4        Makefile.am  missing        package.json       README.md  scripts  src           tags   third_party
autogen.sh  a.wasm          ChangeLog    config.log  configure      configure.ac  debian   fileio_data  libtool     Makefile  Makefile.in  mkinstalldirs  package-lock.json  rpm        snap     subhaobj.txt  tests
admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/sysbench_main$ /home/admin1/Documents/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm --max-threads=64 --dir=fileio_data/ src/sysbench fileio --file-total-size=2G --file-test-mode=rndrw prepare
Attempting to allocate 1064960 bytes of memory...
sysbench 1.1.0-2ca9e3f (using Lua Lua 5.3)

128 files, 16384Kb each, 2048Mb total
Creating files for the test...
Extra file open flags: (none)
FATAL: Can't open file errno = 44 (No such file or directory)

even after providing absolute path , still have the same issue.

Please suggest me what can i rectify to enable fileio operations.

TianlongLiang, 2024-08-21:
I think the fileio benchmark needs to pass a directory to generate the temp files, I can't find that option, so I simple use the current directory, as you can see I pass the option `--dir=.`.

I can successfully run the following commands:

```C
./iwasm -v=5 --max-threads=8 --dir=. sysbench fileio --file-total-size=2M --file-test-mode=rndrw prepare
```

The output is:

```bash
Attempting to allocate 1064960 bytes of memory...
sysbench 1.1.0-2ca9e3f (using Lua Lua 5.3)

128 files, 16Kb each, 2Mb total
Creating files for the test...
Extra file open flags: (none)
Creating file test_file.0
Creating file test_file.1
Creating file test_file.2
Creating file test_file.3
Creating file test_file.4
Creating file test_file.5
Creating file test_file.6
Creating file test_file.7
Creating file test_file.8
Creating file test_file.9
Creating file test_file.10
Creating file test_file.11
Creating file test_file.12
Creating file test_file.13
Creating file test_file.14
Creating file test_file.15
Creating file test_file.16
Creating file test_file.17
Creating file test_file.18
Creating file test_file.19
Creating file test_file.20
Creating file test_file.21
Creating file test_file.22
Creating file test_file.23
Creating file test_file.24
Creating file test_file.25
Creating file test_file.26
Creating file test_file.27
Creating file test_file.28
Creating file test_file.29
Creating file test_file.30
Creating file test_file.31
Creating file test_file.32
Creating file test_file.33
Creating file test_file.34
Creating file test_file.35
Creating file test_file.36
Creating file test_file.37
Creating file test_file.38
Creating file test_file.39
Creating file test_file.40
Creating file test_file.41
Creating file test_file.42
Creating file test_file.43
Creating file test_file.44
Creating file test_file.45
Creating file test_file.46
Creating file test_file.47
Creating file test_file.48
Creating file test_file.49
Creating file test_file.50
Creating file test_file.51
Creating file test_file.52
Creating file test_file.53
Creating file test_file.54
Creating file test_file.55
Creating file test_file.56
Creating file test_file.57
Creating file test_file.58
Creating file test_file.59
Creating file test_file.60
Creating file test_file.61
Creating file test_file.62
Creating file test_file.63
Creating file test_file.64
Creating file test_file.65
Creating file test_file.66
Creating file test_file.67
Creating file test_file.68
Creating file test_file.69
Creating file test_file.70
Creating file test_file.71
Creating file test_file.72
Creating file test_file.73
Creating file test_file.74
Creating file test_file.75
Creating file test_file.76
Creating file test_file.77
Creating file test_file.78
Creating file test_file.79
Creating file test_file.80
Creating file test_file.81
Creating file test_file.82
Creating file test_file.83
Creating file test_file.84
Creating file test_file.85
Creating file test_file.86
Creating file test_file.87
Creating file test_file.88
Creating file test_file.89
Creating file test_file.90
Creating file test_file.91
Creating file test_file.92
Creating file test_file.93
Creating file test_file.94
Creating file test_file.95
Creating file test_file.96
Creating file test_file.97
Creating file test_file.98
Creating file test_file.99
Creating file test_file.100
Creating file test_file.101
Creating file test_file.102
Creating file test_file.103
Creating file test_file.104
Creating file test_file.105
Creating file test_file.106
Creating file test_file.107
Creating file test_file.108
Creating file test_file.109
Creating file test_file.110
Creating file test_file.111
Creating file test_file.112
Creating file test_file.113
Creating file test_file.114
Creating file test_file.115
Creating file test_file.116
Creating file test_file.117
Creating file test_file.118
Creating file test_file.119
Creating file test_file.120
Creating file test_file.121
Creating file test_file.122
Creating file test_file.123
Creating file test_file.124
Creating file test_file.125
Creating file test_file.126
Creating file test_file.127
2097152 bytes written in 6.46 seconds (0.31 MiB/sec).
```

subhakr, 2024-08-21:
Okay thanks for your support.
Only concern is when i am providing path its not working.

TianlongLiang, 2024-08-21:
The path you provided to iwasm is `--dir=fileio_data/`, which gives your wasm program access to that directory. However, the directory you try to use fileio bench to generate temp files is the current directory(`.`), which you don't have access to, that why you have such error like `FATAL: Can't open file errno = 44 (No such file or directory)`

subhakr, 2024-08-21:
Yep, I got it now.
Thank you so much.