wasm-micro-runtime/wasm-micro-runtime #3731: AOT module load failed: resolve symbol __atomic_compare_exchange_4 failed

Kvencc, 2024-08-19:
The following are the details when wamrc is compiled：
wamrc --target=riscv32 --target-abi=ilp32d --size-level=1 --enable-multi-thread --format=aot -o foo.aot foo.asm
Create AoT compiler with:
  target:        riscv32
  target cpu:
  target triple: riscv32-pc-linux-ilp32d
  cpu features:  +d
  opt level:     3
  size level:    1
  output format: AoT file
Compile success, file foo.aot was generated.

Then when I tried to load and run this aot module, this error occurred：

AOT module load failed: resolve symbol __atomic_compare_exchange_4 failed.

I tried to use the same compilation option on other platforms, but only modified the -target option, which seems to work normally.

Do I need to provide implementation for functions like __atomic_compare_exchange_4 and __atomic_store_4 in riscv?

Thank you very much for your reply.

Kvencc, 2024-08-19:
If I don't specify the target, won't wamrc be compiled into the host architecture by default? I have x86_64 here. But I need to run this aot module on the riscv platform.

When I run on the riscv platform if dont specify the target, an error like this will be returned：
AOT module load failed: invalid target bit width, expected 32-bit but got 64-bit

TianlongLiang, 2024-08-19:
So you want to do cross-compiling.

AOT file generated on the x86_64 platform
iwasm ran on the riscv32 platform(is it also compiled on the riscv32 platform or use cross-compile toolchain generated on the x86_64 platform?)

Am I right?

And the error message:
```bash
AOT module load failed: resolve symbol __atomic_compare_exchange_4 failed.
```

refer to the lack of support for atomic intrinsic on the riscv32 platform, I am not very familiar with the risc32 platform, I suspect that it might be some extra compile flag needed to be added when compiling iwasm on the risv32 platform to support atomic operations

Kvencc, 2024-08-19:
Yes, I ported iwasm to my application and used cross-compilation tools to build it on the x86_64 platform.
And the aot file is also generated on the x86 platform.

Can I choose not to use these platform-related features in iwasm? If possible, which options should I specify in the process of compiling iwasm and aot?

Kvencc, 2024-08-19:
I broke the debugging point in the resolve_target_sym function of aot_loader.c, and the following functions returned null:
- __atomic_compare_exchange_4

- __atomic_store_4

- __divsi3

- __mulsi3

- __udivsi3

- __floatdisf

- __floatdidf

So I wonder if some configurations were not done well when I compiled iwasm.

TianlongLiang, 2024-08-20:
If you are not using threads in your wasm app, you can try to remove the option `--enable-multi-thread`. For the other missing symbols, you may try the approach in [this PR](https://github.com/bytecodealliance/wasm-micro-runtime/pull/3733/files) to see whether it helps.

Kvencc, 2024-09-17:
The issue has been solved.

My riscv32 machine supports these features: `-march=rv32imafdc`. So, when I'm using **wamrc** to convert a wasm file to an aot file, I use these options:

```bash
./wamrc --target=riscv32 --target-abi=ilp32d --cpu=generic-rv32 --cpu-features=+a,+m,+c,+f,+d  --format=aot -o foobar.aot foobar.wasm
```

After I added the `--cpu=generic-rv32 --cpu-features=+a,+m,+c,+f,+d` options, the previous problem with the missing symbols didn't recur and the app worked fine!

By the way, if the platform doesn't support these features, I tried to make the following modifications, which also work fine:

```diff
diff --git a/src/aot_reloc_riscv.c b/src/aot_reloc_riscv.c
index b87bb20..c7a36a6 100644
--- a/src/aot_reloc_riscv.c
+++ b/src/aot_reloc_riscv.c
@@ -48,6 +48,14 @@
 #define NEED_SOFT_I64_DIV
 #endif
 
+#ifdef aot_no_ues_mafdc
+#define NEED_SOFT_I32_DIV
+#define NEED_SOFT_I32_MUL
+
+void __atomic_compare_exchange_4();
+void __atomic_store_4();
+#endif
+
 /* clang-format off */
 void __adddf3();
 void __addsf3();
@@ -106,6 +114,10 @@ void __unordsf2();
 static SymbolMap target_sym_map[] = {
     /* clang-format off */
     REG_COMMON_SYMBOLS
+#ifdef aot_no_ues_mafdc
+    REG_SYM(__atomic_compare_exchange_4),
+    REG_SYM(__atomic_store_4),
+#endif
 #ifdef NEED_SOFT_FP
     REG_SYM(__addsf3),
     REG_SYM(__divsf3),
@@ -127,6 +139,8 @@ static SymbolMap target_sym_map[] = {
      * to convert float and long long
      */
     REG_SYM(__floatundisf),
+    REG_SYM(__floatdisf),
+    REG_SYM(__floatdidf),
 #endif
 #ifdef NEED_SOFT_DP
     REG_SYM(__adddf3),
```

TianlongLiang, 2024-09-23:
I am glad that you found out the reason, would you like to make this patch a PR?

Kvencc, 2024-09-23:
Yes, I will create a PR for this patch.

TianlongLiang, 2024-09-24:
That's good to hear, thanks for your contribution!

closed 2024-10-01 by Kvencc