wasm-micro-runtime/wasm-micro-runtime #3747: sysbench mutex workload is taking more execution time with iwasm than native x86_64 execution

subhakr, 2024-08-21:
### Subject of the issue

I have ran both modes like in native x86_64 with gcc and with WAMR runtime.
In native with gcc its taking less execution like below.
```
admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/sysbench_main$ sysbench mutex --threads=60 --time=3 run
sysbench 1.0.20 (using system LuaJIT 2.1.0-beta3)

Running the test with following options:
Number of threads: 60
Initializing random number generator from current time

Initializing worker threads...

Threads started!

General statistics:
    total time:                          2.4519s
    total number of events:              60

Latency (ms):
         min:                                 1426.15
         avg:                                 2212.66
         max:                                 2396.62
         95th percentile:                     2362.72
         sum:                               132759.66

Threads fairness:
    events (avg/stddev):           1.0000/0.00
    execution time (avg/stddev):   2.2127/0.13
```

when it comes to WAMR runtime its taking this much time like below
```
admin1@admin1-VivoBook-ASUSLaptop-X515EA-P1511CEA:~/sysbench_main$ /home/admin1/Documents/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm --max-threads=64 src/sysbench mutex --threads=60 --time=3 run
Attempting to allocate 1064960 bytes of memory...
sysbench 1.1.0-2ca9e3f (using Lua Lua 5.3)

Running the test with following options:
Number of threads: 60
Initializing random number generator from current time

Initializing worker threads...

Threads started!

Throughput:
    events/s (eps):                      0.4328
    time elapsed:                        138.6421s
    total number of events:              60

Latency (ms):
         min:                               129541.83
         avg:                               136631.40
         max:                               138542.29
         95th percentile:                   100000.00
         sum:                              8197884.16

Threads fairness:
    events (avg/stddev):           1.0000/0.00
    execution time (avg/stddev):   136.6314/2.07
```

### Test case

[sysbench_wasm.zip](https://github.com/user-attachments/files/16690298/sysbench_wasm.zip)

### Your environment

*Host OS: Ubuntu 22.04 LTS
*WAMR version: 2.1.1
*CPU architecture: x86_64

### Steps to reproduce

Just run the below command
we can the total elapsed time for this workload
`/home/admin1/Documents/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm --max-threads=64 src/sysbench mutex --threads=60 --time=3 run`
### Expected behavior

It should complete the execution within 3 seconds.

### Actual behavior

Its completing more than 3 seconds which i have passed in commadline

### Extra Info

My worry is i would like to use webassembly to run my applications in future.
if mutex application won't work normally due to this more execution time. It will be problematic in future.
If this mutex workload will work as same as native i will be more happy.

wenyongh, 2024-08-23:
Hi, could you run the workload with AOT mode? You can [build the aot compiler (wamrc)](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/wamr-compiler/README.md) first, and then use it to compile the wasm file into aot file: `wamrc --enable-multi-thread -o sysbench.aot sysbench`, and then run it:
`iwasm --max-threads=64 sysbench.aot mutex --threads=60 --time=3 run`

In my testing, the result of AOT mode is like below:
```bash
Threads fairness:
    events (avg/stddev):           1.0000/0.00
    execution time (avg/stddev):   0.6787/0.09
```
while the result of interpreter mode is:
```bash
Threads fairness:
    events (avg/stddev):           1.0000/0.00
    execution time (avg/stddev):   13.3131/0.52
```
You can see that AOT mode greatly reduces the execution time, hope it meets your requirement.

subhakr, 2024-08-23:
okay i will check it and i have more question about memory workload.
It's giving less results compared to native.
So caan i use this same method for memory workload also.

Thanks in advance.