wasm-micro-runtime/wasm-micro-runtime #3776: CI:

yamt, 2024-09-09:


yamt, 2024-09-09:
@no1wudi any idea?

lum1n0us, 2024-09-09:
I though @wenyongh is working on it too. https://github.com/bytecodealliance/wasm-micro-runtime/pull/3771

wenyongh, 2024-09-09:
I have no good idea yet, at first I think it may be caused by the timeout to wait for compiling wasm to aot is a little small, but it doesn't work even when I increases it. And it may report several errors:
(1) THE FINAL EXCEPTION IS compile wasm to aot failed
(2) THE FINAL EXCEPTION IS argument of type 'NoneType' is not iterable
(3)
Testing(return) f  = -1.797693e+308:f64
THE FINAL EXCEPTION IS Failed:
 Result 0 incorrect:
 expected: '-1.797693e+308:f64'
  got: '-inf:f64'

It fails frequently, I have to re-run it manually many times.

lum1n0us, 2024-09-09:
🤔 I am thinking about "sequential" instead of "parallel" by removing `-P` from https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/10764321632/workflow#L333

no1wudi, 2024-09-09:
> 🤔 I am thinking about "sequential" instead of "parallel" by removing `-P` from https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/10764321632/workflow#L333

Shutting down parallel testing may reduce or even prevent the problem from continuing to occur, but it may lead to excessively long testing times; perhaps we can give it a try.

yamt, 2024-09-09:
> Speed of "sequential" test seems acceptable: #3780
> 
> Maybe we can try it fist
> 
> @wenyongh @lum1n0us @yamt

it's still failing.
https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/10775274630/job/29879549182

wenyongh, 2024-09-10:
It's more like an issue related to riscv32_ilp32f, the CI ran failed most on config "riscv32_ilp32f, fp, -t aot -X" and "riscv32_ilp32f, fp, -t aot":
https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/10775274630/job/29906929175#step:25:7726
https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/10775274630/job/29906623580
https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/10783438239/job/29906934331
https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/10783852668/job/29906610576
https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/10784036686/job/29907065645

wenyongh, 2024-09-10:
How about re-opening #3777 and merging it?

lum1n0us, 2024-09-10:
I agree. It has blocked several PRs.

yamt, 2024-09-10:
> How about re-opening #3777 and merging it?

re-opened and rebased.

yamt, 2024-09-10:
> It's more like an issue related to riscv32_ilp32f, the CI ran failed most on config "riscv32_ilp32f, fp, -t aot -X" and "riscv32_ilp32f, fp, -t aot": [...]

at least riscv64 was failing with similar symptoms too.
i agree riscv32_ilp32f was failing more frequently though.