wasm-micro-runtime/wasm-micro-runtime #3779: Access to call stack when calling native functions

sasq64, 2024-09-09:
#### Feature

When a native function is called, I would like to be able to get the call stack
from the WASM side to see how the call originated.

#### Benefit

If anything goes wrong in the native code, it is often very helpful to see which file and line caused the issue in the web assembly code. For instance if I implement a `load_file()` function, and the file could not be found I can show an exception with information where this call was performed so it can be fixed.

#### Implementation

Since call stack dumps can already be performed, this information should already be available?
Also it is fine if only addresses are provided, the native side can use separate debug info file to look up actual file name and line.

#### Alternatives

...

sasq64, 2024-09-09:
NOTE: Calling `wasm_runtime_dump_call_stack(env)` from the native function does nothing since at that point `module->frames` is empty...
(wasm code built with debug, stack trace enabled, classic interpreter)

loganek, 2024-09-09:
Not sure if that helps, but native function has access to the exec_env structure, so in order to obtain the call stack, you could do something like (haven't tested, so there might be some syntax issues):

```
static void your_native_function(wasm_exec_env_t exec_env)
{
  struct WASMInterpFrame *frame = exec_env->cur_frame; // please note that for AOT this is an instance of `AOTFrame`
  while (frame) {
    // do something with the frame
    frame = frame->prev_frame;
  }
}
```

sasq64, 2024-09-09:
Actually this works;
```cpp
    auto* inst = wasm_runtime_get_module_inst(env);
    auto* module = wasm_runtime_get_module(inst);
    auto ip = (uint32)(cur_frame->prev_frame->ip - ((WASMModule*)module)->load_addr);
```