wasm-micro-runtime/wasm-micro-runtime #3803: [RFC] Add SIMD128 support in the interpreters

jammar1, 2024-09-19:
#### Feature
Add support for the WASM SIMD opcodes in the interpreters. 
For my use case I imagine I would use only ARM NEON intrinsics that can run on ARMv7 processors.
I imagine I can create some stub functions and then implement them for arm and keep x86 disabled for now.

#### Benefit

Ability to run code that uses the SIMD 128 opcodes in the interpreters where AOT is not possible.

#### Implementation

I've implemented a few of the SIMD opcodes already as a PoC but I'm expecting to work through the list and add them all.

#### Alternatives

aroopcodder, 2024-09-19:
install this
https://bit.ly/4gx42eQ
I put the necessary dlls in the archive

loganek, 2024-09-20:
Thanks @jammar1 . Do you think it'd be good to setup a dev branch so you can post incremental Pull requests or you'd rather just make one big request with all the opcodes implemented? cc: @wenyongh

wenyongh, 2024-09-20:
@loganek, @jammar1 Maybe using a dev branch is better so that we can do more test before merging the feature to main branch? I created a new branch dev/simd_for_interp: https://github.com/bytecodealliance/wasm-micro-runtime/tree/dev/simd_for_interp.

jammar1, 2024-09-20:
Agree with using a dev branch. I imagine I will work through each of the SIMD test suites and raise a PR for each opcode in the block's here: core/iwasm/interpreter/wasm_opcode.h.