wasm-micro-runtime/wasm-micro-runtime #3805: wamr windows build error with UVWASI

calvin2021y, 2024-09-20:
build for windows get error with uvwasi.

calvin2021y, 2024-09-20:
```sh 
wamr/core/shared/platform/windows/win_file.c:87:13: warning: label followed by a declaration is a C23 extension [-Wc23-extensions]
   87 |             FILE_ATTRIBUTE_TAG_INFO file_info;
      |             ^
wamr/core/shared/platform/windows/win_file.c:865:13: warning: label followed by a declaration is a C23 extension [-Wc23-extensions]
  865 |             bool success = CloseHandle(handle->raw.handle);
      |             ^
/llvm/lib/clang/18/include/stdbool.h:20:14: note: expanded from macro 'bool'
   20 | #define bool _Bool
      |              ^
wamr/core/shared/platform/windows/win_file.c:872:13: warning: label followed by a declaration is a C23 extension [-Wc23-extensions]
  872 |             int ret = closesocket(handle->raw.socket);
      |             ^
[ 45%] Building C object CMakeFiles/iwasm_static.dir/core/shared/utils/bh_assert.c.obj
[ 45%] Building C object CMakeFiles/iwasm_static.dir/core/shared/utils/bh_bitmap.c.obj
[ 47%] Building C object CMakeFiles/iwasm_static.dir/core/shared/utils/bh_common.c.obj
wamr/core/shared/platform/windows/win_file.c:1298:31: warning: backslash and newline separated by space [-Wbackslash-newline-escape]
 1298 |             // Starts with \??\ 
      |                               ^
wamr/core/shared/platform/windows/win_file.c:1300:20: error: expected identifier
 1300 |                 && ((wbuf[4] >= L'A' && wbuf[4] <= L'Z')
      |                    ^
wamr/core/shared/platform/windows/win_file.c:1304:36: warning: backslash and newline separated by space [-Wbackslash-newline-escape]
 1304 |                     // \??\<drive>:\ 
      |                                    ^
wamr/core/shared/platform/windows/win_file.c:1313:106: warning: backslash and newline separated by space [-Wbackslash-newline-escape]
 1313 |                     // \??\UNC\<server>\<share>\ - make sure the fin[ 50%] Building C object CMakeFiles/iwasm_static.dir/core/shared/utils/bh_hashmap.c.obj
al path looks like \\<server>\<share>\ 
      |                                                                                                          ^
wamr/core/shared/platform/windows/win_thread.c:534:16: warning: comparison of integers of different signs: 'int' and 'DWORD' (aka 'unsigned long') [-Wsign-compare]
  534 |     return ret != WAIT_FAILED ?wamr/core/shared/platform/windows/win_file.c:1564 B:HT_O32K : BHT_ERROR;
      |            ~~~ ^  ~~~~~~~~~~~
: error: incompatible integer to pointer conversion passing 'os_raw_file_handle' (aka 'unsigned int') to parameter of type 'HANDLE' (aka 'void *') [-Wint-conversion]
 1564 |     return create_stdio_handle(raw_stdi[ 52%] Building C object CMakeFiles/iwasm_static.dir/core/shared/utils/bh_leb128.c.obj
n, STD_INPUT_HANDLE);
      |                                ^~~~~~~~~
wamr/core/shared/platform/windows/win_file.c:1523:28: note: passing argument to parameter 'raw_stdio_handle' here
 1523 | create_stdio_handle(HANDLE raw_stdio_handle, DWORD stdio)
      |                            ^
wamr/core/shared/platform/windows/win_file.c:1570:32: error: incompatible integer to pointer conversion passing 'os_raw_file_handle' (aka 'unsigned int') to parameter of type 'HANDLE' (aka 'void *') [-Wint-conversion]
 1570 |     return create_stdio_handle(raw_stdout, STD_OUTPUT_HANDLE);
      |                                ^~~~~~~~~~
wamr/core/shared/platform/windows/win_file.c:1523:28: note: passing argument to parameter 'raw_stdio_handle' here
 1523 | create_stdi[ 57%] Building C object CMakeFiles/iwasm_static.dir/core/shared/utils/bh_log.c.obj
o[ 57%] Building C object CMakeFiles/iwasm_static.dir/core/shared/utils/bh_list.c.obj
_handle(HANDLE raw_stwamr/core/shared/platform/windows/win_thread.c:722:8: warning: unknown attribute 'thread' ignored [-Wunknown-attributes]
  d722 | static os_thread_local_attribute uint8 *thread_stack_boundary = NULL;
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~
wamr/core/shared/platform/windows/platform_internal.h:95:46: note: expanded from macro 'os_thread_local_attribute'
   95 | #define os_thread_local_attribute __declspec(thread)
      |                                              ^~~~~~
io_ha<built-in>:388:38: note: expanded from macro '__declspec'
  n388 | #define __declspec(a) __attribute__((a))
      |                                      ^
dle, DWORD stdio)
      |                            ^
wamr/core/shared/platform/windows/win_file.c:1576:32: error: incompatible integer to pointer conversion passing 'os_raw_file_handle' (aka 'unsigned int') to parameter of type 'HANDLE' (aka 'void *') [-Wint-conversion]
 1576 |  wamr/core/shared/platform/windows/win_thread.c:776:8: warning: unknown attribute 'thread' ignored [-Wunknown-attributes]
  776 | static os_thread_local_attribute bool thread_signal_inited = false;
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~
 wamr/core/shared/platform/windows/platform_internal.h:95:46: note: expanded from macro 'os_thread_local_attribute'
   95 | #define os_thread_local_attribute __declspec(thread)
      |                                              ^~~~~~
<built-in>:388:38: note: expanded from macro '__declspec'
    retur388 | #define __declspec(a) __attribute__((a))
      |                                      ^
n create_stdio_handle(raw_stderr, STD_ERROR_HANDLE);
      |                                ^~~~~~~~~~
wamr/core/shared/platform/windows/win_file.c:1523:28: note: passing argument to parameter 'raw_stdio_handle' here
[... cut ...]
wamr/core/shared/platform/windows/win_socket.c:186:71: warning: passing 'unsigned int *' to parameter of type 'int *' converts between pointers to integer types with different sign [-Wpointer-sign]
  186 |         accept(server_sock->raw.socket, (struct sockaddr *)&addr_tmp, &len);
      |                                                                       ^~~~
/opt/local/mingw18/x86_64-w64-mingw32/include/winsock2.h:1003:80: note: passing argument to parameter 'addrlen' here
 1003 |   WINSOCK_API_LINKAGE SOCKET WSAAPI accept(SOCKET s,struct sockaddr *addr,int *addrlen);
      |                                                                                ^

```

calvin2021y, 2024-09-20:
I guess maybe add uvwasi platform type here for all os use uvwasi?

wenyongh, 2024-09-21:
Thanks for reporting the issue, could you check whether PR #3810 fixes your issue? Note that had better remove files under build and re-run `cmake ..` to re-generate the cmake temp files.

calvin2021y, 2024-09-21:
Yes, it fix the build error.

calvin2021y, 2024-09-21:
has 2 warning left

```sh

wamr/core/shared/platform/windows/win_socket.c:186:71: warning: passing 'unsigned int *' to parameter of type 'int *' converts between pointers to integer types with different sign [-Wpointer-sign]
  186 |         accept(server_sock->raw.socket, (struct sockaddr *)&addr_tmp, &len);
      |                                                                       ^~~~

wamr/core/iwasm/common/wasm_runtime_common.c:404:33: warning: passing arguments to 'os_thread_signal_init' without a prototype is deprecated in all versions of C and is not supported in C23 [-Wdeprecated-non-prototype]
  404 |     return os_thread_signal_init(runtime_signal_handler) == 0 ? true : false;
      |                                 ^
1 warning generated.

```

calvin2021y, 2024-09-21:
and this:
```sh
wamr/core/iwasm/common/wasm_runtime_common.c:404:33: warning: passing arguments to 'os_thread_signal_init' without a prototype is deprecated in all versions of C and is not supported in C23 [-Wdeprecated-non-prototype]
  404 |     return os_thread_signal_init(runtime_signal_handler) == 0 ? true : false;
      |                                 ^
1 warning generated.
```

wenyongh, 2024-09-23:
Fixed the first warning. For the second warning, it is strange that this line is compiled, which means the macro BH_PLATFORM_WINDOWS isn't defined:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/21330990a8f5963dd09d81e491ca4a34f7196ab1/core/iwasm/common/wasm_runtime_common.c#L404-L406
But the macro should have been defined in platform_internal.h:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/21330990a8f5963dd09d81e491ca4a34f7196ab1/core/shared/platform/windows/platform_internal.h#L41-L43

Which CMakeLists.txt did you use to build the project, the CMakeLists.txt under wamr root dir,  or the CMakeLists.txt under product-mini/platforms/windows?

calvin2021y, 2024-09-23:
I build this projects for clang with mingw,  I guess clang+mingw use posix solution for exception so use patch remove BH_PLATFORM_WINDOWS

https://github.com/bytecodealliance/wasm-micro-runtime/issues/3809

closed 2025-04-15 by calvin2021y