wasm-micro-runtime/wasm-micro-runtime #3816: API for custom native stack registration

mkokryashkin, 2024-09-26:
#### Feature
Currently my application runs WASM modules withing custom C fibers. These fibers have their own stacks, different from their thread stack. WAMR, on the other hand, heavilly relies on pthread API for native stack boundary checking.
It would be nice to have a handle like
```
wasm_runtime_register_native_stack_boundary(uint8_t *stack_boundary)
```

So it is possible to register your own native stack boundary and use WAMR within a fiber.

#### Implementation
We just need to set `exec_env->native_stack_boundary` to a custom value.

wenyongh, 2024-10-15:
Hi, it is interested to support fibers. Currently the `exec_env->native_stack_boundary` is updated in each time `wasm_call_function/aot_call_function`, which calls `wasm_exec_env_set_thread_info` to get native stack boundary from os and update it. Per my understanding, you want to use the registered native stack boundary in `wasm_call_function/aot_call_function`? Seems it is better to store the registered native stack boundary in a new field of exec_env and use it if it is not NULL in `wasm_exec_env_set_thread_info`?

mkokryashkin, 2024-10-15:
> Seems it is better to store the registered native stack boundary in a new field of exec_env and use it if it is not NULL in `wasm_exec_env_set_thread_info`?

Yes, that is a great option and it adds a lot of flexibility.

wenyongh, 2024-10-16:
I uploaded PR #3862, could you check whether it resolves your issue? You can call wasm_runtime_set_native_stack_boundary to set the native stack boundary to exec_env.

mkokryashkin, 2024-10-16:
It works nicely, thank you very much!

wenyongh, 2024-10-23:
Close this issue since https://github.com/bytecodealliance/wasm-micro-runtime/pull/3862 was merged.

closed 2024-10-23 by wenyongh