wasm-micro-runtime/wasm-micro-runtime #3829: Please publish release artefacts with GC and EH support

bashor, 2024-09-30:
#### Feature

To get GC and EH support, right now, we need to build WAMR from sources, which complicates 
* trying WAMR with Kotlin/Wasm projects for users
* using WAMR in our [examples](https://kotl.in/wasm-examples/)
* using WAMR in our compiler tests

Ideally, we need builds for [Linux | macOS | Windows] * [x86 | Arm] (64-bit versions probably would be enough), though I'm not quite sure about Linux Arm, macOS x86.

#### Benefit

Having such binaries in releases may remove/reduce barriers to trying WAMR on wasm binaries requiring GC and EH support.
Also, in this case, it would be simpler for us to use WAMR in our compiler tests and examples

#### Alternatives

Someone else could do it, but official binaries are likely more trustworthy. 
Also, it's simpler to keep such builds up to date inside the WAMR project.

TianlongLiang, 2024-10-18:
Thanks for your suggestion, I modified the release CI to add the builds for x86 [Linux | macOS | Windows], and on Windows and Linux, test your wasi example, they can run successfully.

bashor, 2024-10-18:
🙌 Great! Thanks!

bashor, 2024-10-18:
How about Arm+macOS?

TianlongLiang, 2024-10-21:
I will look into arm macOS and it will probably be ready in the release after the next(next is 2.2.0, which will be released soon, so maybe release binaries can have arm+macOS in 2.2.1 or so), I want to test it more on the arm macOS but I don't have the environment right now.

bashor, 2024-10-21:
@TianlongLiang awesome! We are looking forward to it!

wenyongh, 2024-10-23:
@bashor https://github.com/bytecodealliance/wasm-micro-runtime/releases/tag/WAMR-2.2.0.

bashor, 2024-10-23:
@wenyongh super, thanks for the update!

lum1n0us, 2026-01-30:
@bashor  Due to [this pull request](https://github.com/bytecodealliance/wasm-micro-runtime/pull/4817), we are considering not releasing gc-eh in the future. I'm not sure if this will impact your work, so please let me know.