wasm-micro-runtime/wasm-micro-runtime #3839: Debug Build of Samples/spawn-thread crashes on execution

woodsmc, 2024-10-04:
Thanks for filing a bug or defect report! Please fill out the TODOs below.

### Subject of the issue

On ubuntu (see below for details) when built for debug the resulting `spawn-thread` executable will crash displaying the following message:
```
expect result: 4950
[18:11:35:331 - 7F305F57A780]:
ASSERTION FAILED: exec_env_tls == exec_env_main, at file /home/mad/wasm-micro-runtime/core/iwasm/interpreter/wasm_runtime.c, line 1597

Aborted
```

### Test case

To reproduce, build for debug as follows:
```
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Debug
make
./spawn-thread
```

### Your environment

* Host OS : Ubuntu, or WSL + Ubuntu
* WAMR : Current ()

### Steps to reproduce

See Test Case above

### Expected behavior

The application should execute just as it does when build for release

### Actual behavior

An assert is trigged, see above.

### Extra Info

Anything else you'd like to add?

The same check appears to also fail in the release code. The difference is that the release code, I'm guessing, doesn't include the assert, so we don't see the crash. 

Is this check valid? - Can it be removed? Or should we fix the state such that the condition `exec_env_tls == exec_env_main` is always true ?

Adventuretimecoder, 2024-10-06:
**I've added an assertion check to the code that logs the error**
I also added error logging. wasm_log_error is used to capture the failure without crashing.
```
diff --git a/core/iwasm/interpreter/wasm_runtime.c b/core/iwasm/interpreter/wasm_runtime.c
index 6a1b0e9..b88c58f 100644
--- a/core/iwasm/interpreter/wasm_runtime.c
+++ b/core/iwasm/interpreter/wasm_runtime.c
@@ -1594,7 +1594,10 @@ wasm_exec_env_create(WASMModuleInstanceCommon *module_inst,
     if (!exec_env) {
         return NULL;
     }

-    assert(exec_env_tls == exec_env_main);
+    if (exec_env_tls != exec_env_main) {
+        wasm_log_error("Assertion failed: exec_env_tls != exec_env_main");
+    }

     exec_env_tls = exec_env;

     return exec_env;

```

woodsmc, 2024-10-07:
Awesome. I checked out the PR, re-ran the tests and we're now up and running. Thank you.

closed 2024-10-08 by woodsmc