wasm-micro-runtime/wasm-micro-runtime #3842: Windows Builds Missing from Release Assets

AjaniBilby, 2024-10-06:
I assume there is reason for this based on some of the open issues of people trying to build `iwasm` themselves, but I couldn't find any exlicit reason why there is no windows build in the release assets.

The only reason I haven't just built it myself is because my windows install is messed up and my path keeps lossing things, and my linux machine is RMA...

lum1n0us, 2024-10-07:
This is just what I think. Anyone, @wenyongh, is welcome to add their thoughts.

One reason might be that there aren't many developers in the community who are familiar with Windows for development. I think this might be because most of the WAMR use cases we know about don't use Windows. Anyway, not having enough experience makes the porting job go slower and leads to the issue you talked about. We're open to any help, especially in this area. 🤝

AjaniBilby, 2024-10-07:
I'm not directly that interested in WAMR itself, but rather just as a runtime for wasm.
And from my own benchmark comparing `wasmer` with hand rolled wat to native C for a basic http ping server it had a 50% performance loss, and going through their rust code base is like seven different layers of Rust abstraction to get to any thing actually executing at an OS or wasm level.

AjaniBilby, 2024-10-07:
It seems that wamr is the only runtime that has both GC and berkley sockets, so I guess I'll need to start getting into building it.

wenyongh, 2024-10-07:
Agree, it would be great if there are windows binaries in the release assets, e.g. iwasm.exe and wamrc.exe. But it seems that @AjaniBilby wants iwasm.exe with GC and socket support, and other developer also hopes to release iwasm with GC support, see #3829, so my suggestion is to release: (1) iwasm without GC support for macos, ubuntu and windows, (2) iwasm with GC support for macos, ubuntu and windows, (3) wamrc for macos, ubuntu and windows.

AjaniBilby, 2024-10-10:
It looks like @wenyongh and others have already done most of what I would have been able to [contribute already](https://github.com/bytecodealliance/wasm-micro-runtime/commits/main/core/shared/platform/windows) 🫡

Also the latest release of wabt isn't able to convert the example in the proposal overview (adjusted to the latest spec) from wat to wasm. So it definetly is a bleeding edge feature, and I'll just rely on V8 based runtimes for now iWasm is ready 🙇

closed 2024-10-10 by AjaniBilby